theseus-os/Theseus #322: error cfg(doctest) is experimental and subject to change while installing xargo

omac777, 2020-11-07:
PROBLEM

When only the nightly rust compiler is installed, you will encounter the following error.
```
error[E0658]: `cfg(doctest)` is experimental and subject to change
  --> /home/david/.cargo/registry/src/github.com-1ecc6299db9ec823/remove_dir_all-0.5.3/src/lib.rs:12:7
   |
12 | #[cfg(doctest)]
   |       ^^^^^^^
   |
   = note: for more information, see https://github.com/rust-lang/rust/issues/62210
   = help: add `#![feature(cfg_doctest)]` to the crate attributes to enable
```

SOLUTION

```
rustup toolchain install stable
cargo +stable install --vers 0.3.17 xargo
```

kevinaboos, 2020-11-07:
Thanks for the report & fix! This is a known issue that I've yet to get around to, it'd also be resolved when we move to a more recent compiler version, but it's good to have a fix now anyway. I'll make the changes to the README and related build files.

kevinaboos, 2020-11-13:
Fixed in PR #329, closed by commit 31937ac

closed 2020-11-13 by kevinaboos