theseus-os/Theseus #325: build_docker script crashes due to logname error

AnActualEmerald, 2020-11-08:
I was trying to get up and running using docker and encountered an issue where the `docker build` command in the build_docker.sh script was crashing because the user arg wasn't being set. I checked what the output of logname is on my machine and all I got back was `logname: no login name`. I swapped out `logname` for `whoami` in the script and it seems to have built just fine. 

I'm not sure if this is just something weird about how I have my system configured, which is admittedly pretty likely, and I'm certainly sure this isn't a high priority issue, but I figured I'd bring it up for thoroughness' sake. I'm running Linux Mint 19.3, fish shell 3.1.2, though I double checked with bash as well and got the same result

kevinaboos, 2020-11-08:
Thanks for the report! Long story short, I wasn't sure if all Linux distributions had the `whoami` executable or had the `LOGNAME` env var defined. I'll add the use of `whoami` as a backup now. 

Out of curiosity, do you have any other related env vars defined to be your username on your system?

AnActualEmerald, 2020-11-08:
`LOGNAME` and `USER` are both defined as my username, I don't see any others looking at the output of `env`

kevinaboos, 2020-11-08:
Ok so you're just missing the logname executable then. Thanks for the clarification!

commit 2b693e2d (2020-11-10): Use 'id' instead of 'logname' in docker scripts. Closes #325.
closed 2020-11-10 by kevinaboos