theseus-os/Theseus #335: How is overlapping MappedPages.as_type/as_type_mut safe?

tv42, 2020-12-30:
I may be missing something very fundamental.

It seems if I use MappedPages, and get two different data types pointing to the same area (I see nothing in the code preventing that), at least one of which is via `as_type_mut`, I can treat a pointer as integer by aliasing and start doing arbitrary unsafe things.

tv42, 2020-12-30:
Similarly, it seems all cells (including non-OS "applications") can get MappedPages for arbitrary physical addresses at any time, regardless of whether something else has already mapped those pages? Just pass an integer to PhysicalAddress.new, make a PhysicalMemoryArea, and you can map pages, right?

This is starting to just seem like a part of the Single Privilege Level design, but I would have expected after all that talk about safeness, this would be at least mentioned as a caveat. It's trivial to be hostile, and it's still reasonably simple to shoot yourself in the foot.

kevinaboos, 2021-01-04:
Thanks for the questions! You are correct about the current version of the code in this public repo, the version discussed in our OSDI paper does not have this issue because there is only one way to obtain a set of physical frames usable for memory mapping purposes: via the `AllocatedFrames` type from the a frame allocator. There were some small bugs in this version so they haven't yet been merged into the main public repo, but that should happen soon. Using those types along with `AllocatedPages` ensures that every mapping is bijective and cannot overlap. Also, the `PhysicalMemoryArea` is part of an older design that is being phased out and isn't really used for mappings directly, but as part of the frame allocator.

Your other question about acquiring multiple references to a single `MappedPages` instance is addressed by Rust's "aliasing XOR mutability" rule, in which you can only obtain a single mutable reference to a given object at one time, and not multiple mutable references nor a mutable reference and one or more immutable references. Currently the `MappedPages` types does not prevent one from obtaining multiple _immutable_ references to its underlying memory region, as that does not present safety issues AFAICT.

Limiting applications' access to functionality is a solved problem in the SAS/SPL OS design space, and in our case would be realized by allowing applications to depend on only the functions and types publicly re-exported by a `libtheseus` library, which is in the works.  The way that applications currently access kernel-classified crate functionality is just to show a simple proof of concept as to how applications might be efficiently realized; they're not limited in any way, but I agree that model isn't "production-ready", per se.

tv42, 2021-01-04:
> "aliasing XOR mutability"

I'm definitely no Rust expert, but I am aware of the general rule. I think I'm missing something where that translates from MappedPage level to the individual types inside the page, at the given offset. Oh except if.. you follow a restriction I hadn't thought of, where the lifetime of the MappedPage really does become the lifetime of a single type inside the page, not avoided via unsafe.

Updated understanding, hopefully more correct: You cannot simultaneously hold both a mutable type at one offset and anything else mutable or immutable even at non-overlapping offsets. You only get to "spend the mut MappedPage once". That is, Rust lifetimes get enforced on the whole page level, not at the individual types.

I was expecting to see some sort of access control, either via the ELF boundaries or by passing some otherwise unaccessible state to the startup function, not surprised at all that that's the plan -- would have been nice to see that mentioned up front though. That gets kinda interesting with the SPL angle, though. Some animals inside the single level animal farm are more equal than others..

Neat work!

kevinaboos, 2021-01-04:
Yes, your "Updated understanding" paragraph is right. Sorry if I didn't make that clear in the `MappedPages` docs, I should do so; hopefully it's clearer in our paper. The fact that `MappedPages`'s access methods only allow the caller to obtained a reference to an overlaid type `T` and not a distinct owned instance of that type `T` is precisely how we ensure this lifetime relationship persists. I believe that's one of the most important invariants therein. 

Access control is indeed a fascinating concept to explore within the OS design space, but I felt that it had largely been solved in various ways for many points in that space, monolithic, microkernel, and SAS/SPL systems. Our research group hasn't yet attempt to make any new contribution there, so it's not a part of this work (at least not yet).

closed 2021-01-05 by kevinaboos