theseus-os/Theseus #339: Question: where can I read about other experimental os

imvetri, 2021-01-12:
I'm researching for possible ways to integrate neural network as part of the operating systems. Idea is to save the computation for repeated instruction sets and reuse them for future. I'm looking for a wide range of fantasy OS designs and other experimental os designs.

kevinaboos, 2021-01-12:
Hi @imvetri, thanks for the question! I'm not really sure if this is the best venue for a discussion about that, but due to a lack of other options I'll list a few of my favorite ones here from the OS research literature, and hopefully that'll help to kickstart your learning experience!

Personally I have almost no knowledge of machine learning so I don't have any ideas how it could be integrated into OS design. 

My favorite OSes from the literature or with neat designs (just a quick, incomplete list):
* Singularity, by Microsoft Research. Explored SAS/SPL designs, wasn't the first but did it very deeply. 
* Mirage OS, a unikernel written in OCaml
* Flexible library OSes:  Exokernel and Drawbridge OS
* seL4 and CertiKOS:  deep works on OS verification
* Proteos, a major extension to MINIX 3 for safe live update of userspace OS servers in a microkernel environment
* SPIN, an older system that explored safely executing custom extensions in the kernel itself
* CuriOS:  a microkernel that recognized the problems of state-related fate sharing among microkernel userspace services and their clients
* Genode, a capability-oriented system with a really cool tree-like OS process structure
* I like a lot of Michael Swift & Brian Bershad's works on OS reliability, like Nooks, Shadow Drivers. Also see George Candea's work called Microreboot

There's so many out there to list, but I'll stop here for now. Hope you enjoy diving into the world of OS design!

imvetri, 2021-01-13:
@kevinaboos thank you very much for the help.

closed 2021-01-13 by imvetri