theseus-os/Theseus #341: Makefile don't work with CARGO_TARGET_DIR

12101111, 2021-01-18:
When CARGO_TARGET_DIR is set, makefile can't find ./target/x86_64-theseus/release/libnano_core.a:

```
make: *** No rule to make target '/home/han/Project/Theseus/target/x86_64-theseus/release/libnano_core.a', needed by '/home/han/Project/Theseus/build/nano_core/nano_core-x86_64.bin'.  Stop.
```

kevinaboos, 2021-01-18:
Thanks for the report! Yeah, the build commands don't support a different target directory than the default one called `target`. 

I can look into making the changes myself, but it's a low-priority change. I'd be happy to accept a PR that fixes it, which would probably just entail changing everywhere that uses a hard-coded "target" directory in the makefiles.

kevinaboos, 2021-07-01:
Closing due to inactivity and little benefit from doing so.

closed 2021-07-01 by kevinaboos