theseus-os/Theseus #348: Cannot run cargo test?

pingiun, 2021-01-22:
I might be doing something wrong, but when running `cargo test` in the workspace root I get the included error. Can tests only be run from individual crates?

```
error: duplicate lang item in crate `std` (which `test` depends on): `panic_impl`.
  |
  = note: the lang item is first defined in crate `panic_entry` (which `nano_core` depends on)
  = note: first definition in `panic_entry` loaded from /Users/jelle/src/Theseus/target/debug/deps/libpanic_entry-b4ae55420443d23e.rlib
  = note: second definition in `std` loaded from /Users/jelle/.rustup/toolchains/nightly-2020-11-10-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libstd-ff8f19287472ab25.dylib, /Users/jelle/.rustup/toolchains/nightly-2020-11-10-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libstd-ff8f19287472ab25.rlib

error: duplicate lang item in crate `std` (which `test` depends on): `oom`.
  |
  = note: the lang item is first defined in crate `panic_entry` (which `nano_core` depends on)
  = note: first definition in `panic_entry` loaded from /Users/jelle/src/Theseus/target/debug/deps/libpanic_entry-b4ae55420443d23e.rlib
  = note: second definition in `std` loaded from /Users/jelle/.rustup/toolchains/nightly-2020-11-10-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libstd-ff8f19287472ab25.dylib, /Users/jelle/.rustup/toolchains/nightly-2020-11-10-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libstd-ff8f19287472ab25.rlib

error: duplicate lang item in crate `panic_unwind` (which `std` depends on): `eh_personality`.
  |
  = note: the lang item is first defined in crate `panic_entry` (which `nano_core` depends on)
  = note: first definition in `panic_entry` loaded from /Users/jelle/src/Theseus/target/debug/deps/libpanic_entry-b4ae55420443d23e.rlib
  = note: second definition in `panic_unwind` loaded from /Users/jelle/.rustup/toolchains/nightly-2020-11-10-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libpanic_unwind-e3d969bfc9686736.rlib

error: aborting due to 3 previous errors

error: could not compile `nano_core`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

kevinaboos, 2021-01-22:
I haven't used cargo test much in this project, and when we did, it was on individual crates, yes.  I think previously we had gated some of those Rust lang items with something like `#[cfg(not(test))]`, so perhaps that would fix this issue for you? 

I don't recall if cargo supports running all tests on virtual workspaces, if it does it would likely require the `--all` flag. In any case, most Theseus functionality can't really be tested via `cargo test` due to its hardware-specific and platform-specific nature. 

I know that Philipp Oppermann has spent some time developing a QEMU-based testing infrastructure for his blog os project. We could likely use this infrastructure too, in a limited capacity.

closed 2021-03-03 by kevinaboos

amab8901, 2022-12-02:
> I haven't used cargo test much in this project, and when we did, it was on individual crates, yes. I think previously we had gated some of those Rust lang items with something like [...]

How does hardware and platform specificity make us unable to implement `cargo test` for the entire OS?

Also, do you have solid proof that it's impossible to implement `cargo test` for entire OS? 

I'm a bit skeptical of the idea that it's impossible. Many times, people say that something is impossible just because they couldn't figure out how to do it, rather than it actually being impossible. For example when people said it was impossible to reuse rockets and Elon Musk did it anyway.

kevinaboos, 2022-12-02:
So the way that `cargo test` works by default is that it compiles a binary that runs as a userspace application for a specific host platform. I don't think (and did not say) that it's impossible to do so for many test cases, just that we haven't devoted sufficient time to getting it to work, and that Philipp had implemented some basic QEMU-based test runners that could serve as a starting point for us to build a test infrastructure for our target-specific code.

However, _some_ test cases are indeed very difficulty (nigh impossible) to test using a QEMU-based infra, primarily because QEMU doesn't support that specific hardware feature. For example, we could not possibly use QEMU to test mellanox NICs because there is no emulated instance of them. There are many other examples, but I don't think that should dissuade someone from working on QEMU test infra for Theseus.