theseus-os/Theseus #349: Ensure all dependencies are owned by `theseus-os` org or are from crates.io

kevinaboos, 2021-01-22:
Move the following repositories to be under the ownership of the `theseus-os` org:
* `kevinaboos/xmas-elf`  (likely not needed any more?)
* `kevinaboos/zero`
* `kevinaboos/getopts`
* `kevinaboos/irq_safety`
* `kevinaboos/stable_deref_trait`
* `kevinaboos/owning-ref-rs`

Fork `m-labs/smoltcp` too, while we're at it.

pingiun, 2021-02-03:
Can you explain the argument against having them in tree? In #347 you say this is a size issue, but I don't see the problem in making the repo somewhat bigger, as you need to pull these crates in anyway.

I think it would benefit development productivity to have the crates in tree, as Theseus needs to be build as one "thing" anyway and having all the subelements at a click on the file seems useful to me.

kevinaboos, 2021-02-11:
Hi, apologies for not seeing this earlier. 

Beyond the potential size problem, it's more of a logical organization issue. In my view, repositories that are considered standalone libraries, especially those that originated from a third party, should be maintained as separate artifacts since they don't need to be part of the kernel. In other words, if a library crate does not _need_ to be specialized for Theseus specifically (most of these are just specialized for `no_std`), then it should be kept separate to allow others to reuse it in non-Theseus contexts.  None of these libraries need to depend on code that exists in Theseus's `kernel` directory, therefore they can exist as independent libraries.

Additionally, it maintains the lineage more clearly, i.e., the fact that our fork of `stable_deref_trait` comes from `Storyyeller/stable_deref_trait`. Clarifies the provenance of licensing as well.

As an aside, there are benefits to vendoring, e.g., weathering a failure of github or crates.io, but we should either vendor _every_ third-party crate or none of them, rather than just select random libraries.

kevinaboos, 2021-03-04:
Closed by 75c152f

closed 2021-03-04 by kevinaboos