theseus-os/Theseus #363: failed to get `x86_64` as a dependency of package memory v0.1.0

bobd988, 2021-03-27:
Hi,

I was trying to run this on my mac Mojave. Following the MacOS  instructions to build. The python env is default 2.7. The rust is 1.49.0-nightly. 

./scripts/mac_os_build_setup.sh
 ....
==> Summary
🍺  /usr/local/Cellar/python@3.8/3.8.8_1: 4,376 files, 68.8MB
Removing: /usr/local/Cellar/python@3.8/3.8.2... (4,144 files, 63.2MB)
Error: Directory not empty @ dir_s_rmdir - /usr/local/Cellar/python@3.8/3.8.2
Then I followed this 
rm -rf $HOME/theseus_tools_src $HOME/theseus_tools_opt /opt/local/bin/make

Build again no errors. Next I already has rust so skip the installation step. After that from project root folder 

make run

Here are logs 

=================== BUILDING ALL CRATES ===================
	 TARGET: "x86_64-theseus"
	 KERNEL_PREFIX: "k#"
	 APP_PREFIX: "a#"
	 THESEUS_CONFIG (before build.rs script): ""
RUST_TARGET_PATH="/Users/demo/Theseus/cfg" RUSTFLAGS="--emit=obj -C debuginfo=2 -C code-model=large -C relocation-model=static -D unused-must-use -Z merge-functions=disabled -Z share-generics=no" cargo build   -Z unstable-options -Z build-std=core,alloc -Z build-std-features=compiler-builtins-mem  --all --target x86_64-theseus
    Updating git repository `https://github.com/theseus-os/downcast-rs`
    Updating git repository `https://github.com/theseus-os/getopts`
    Updating git repository `https://github.com/theseus-os/qp-trie-rs`
    Updating git repository `https://github.com/m-labs/smoltcp`
    Updating git repository `https://github.com/theseus-os/volatile`
    Updating git repository `https://github.com/theseus-os/irq_safety`
error: failed to get `x86_64` as a dependency of package `memory v0.1.0 (/Users/demo/Theseus/kernel/memory)`

Caused by:
  failed to load source for dependency `x86_64`

Caused by:
  Unable to update /Users/demo/Theseus/libs/x86_64

Caused by:
  failed to read `/Users/demo/Theseus/libs/x86_64/Cargo.toml`

kevinaboos, 2021-03-29:
hey @bobd988, thanks so much for this report! Unfortunately I don't have a macOS machine to test this on, so I may not be able to reproduce this, but it does appear that this issue may not be related to macOS at all. 

I'm not sure why this is the case, but I will attempt to repro it on a blank slate VM and get back to you.

kevinaboos, 2021-03-29:
oh wait, it just occurred to me that you're likely missing the git submodules. Do you have a `libs/x86_64` subdirectory in your local repo? If not, you need to either
* use the `git clone --recursive ...` argument when checking out the repo for the first time, or
* grab all the submodules, as specified in the [README](https://github.com/theseus-os/Theseus#building-theseus)

If that still doesn't fix things, do let me know so I can look into it further. Thanks!

kevinaboos, 2021-04-26:
Closing due to inactivity. Likely solved by checking out the git submodules.

closed 2021-04-26 by kevinaboos