theseus-os/Theseus #367: Failed in MacOS catalina

virtuehive, 2021-03-30:
Hi . I've tried several times in MacOS . rustc (nightly-2020-11-10) and xargo (0.3.13) are already match with requirement . seems like failed during run/boot process .
Finished release [optimized] target(s) in 1.22s
/Users/newmac/Theseus/kernel/nano_core/src/boot/arch_x86_64/boot.asm:446: error: symbol `INITIAL_STACK_SIZE' not defined
make: *** [Makefile:274: /Users/Loomycute/Theseus/build/nano_core/boot/x86_64/boot.o] Error 1

kevinaboos, 2021-03-30:
Hi @virtuehive, thanks very much for the report! I don't have a macOS machine to test with, but I have a few ideas.

It appears to be a problem with the linker or perhaps the assembler on macOS, in that it differs from the one on Linux. The `INITIAL_STACK_SIZE` symbol isn't being defined properly, which is odd. This difference was likely exposed by [a recent commit from a week ago](https://github.com/theseus-os/Theseus/commit/11ea84f5ff8203039f092466754581a4ac248e68). 

Can you try an older commit, such as c44837d, to see if the problem is still there? Thanks!

virtuehive, 2021-05-09:
Same problem :( 

Finished release [optimized] target(s) in 1.02s
/Users/Loomycute/Theseus/kernel/nano_core/src/boot/arch_x86_64/boot.asm:445: error: symbol `INITIAL_STACK_SIZE' not defined
make: *** [Makefile:274: /Users/Loomycute/Theseus/build/nano_core/boot/x86_64/boot.o] Error 1

kevinaboos, 2021-05-09:
Huh, that's really weird. Thanks for following up! 

The `INITIAL_STACK_SIZE` thing didn't even exist in commits earlier than c44837d so I don't see how that would possibly throw an error about that.

I'm not sure about the exact solution, but perhaps you could try inserting a line into the top of `boot.asm` that defines the `INITIAL_STACK_SIZE` to a value like 32? Or remove the `%ifndef` conditional on line 15 and its corresponding `endif` on line 21?

Thanks again for investigating this! Wish I could be of more help but I don't have a Mac.

kevinaboos, 2021-05-09:
Just to double check, you're not defining the variable `INITIAL_STACK_SIZE` on the command line or anything like that?

virtuehive, 2021-05-10:
no . I didn't define INITIAL_STACK_SIZE on the command line .  all is running well when I write initial stack size manually at line 445 . very cool OS and pretty big potential

kevinaboos, 2021-05-10:
@virtuehive  Thanks for confirming that. It seems like we need to find a different way to define `INITIAL_STACK_SIZE` that's compatible with the assembler toolchains on macOS. If you have the time/ability to come up with another solution to this, I'd greatly appreciate it and would accept a PR to fix it.

virtuehive, 2021-05-10:
What I am starting to like theseus is on android (kotlin + NDK's C)  and ios (Swift +Objective C) apps , we still code the apps on (mostly) X86 platform which is different world . Hopely with cells concept we just need rust-lang  to build the apps  within the OS , even re-built the OS customized for special purposes not only in X86 , but maybe in ARM or RISC-V.  I'm focused on that  :)

aghoneim92, 2021-05-14:
I think I've figured out the problem. It seems to be happening because the code on line 17 is %elif, not %else. When I changed it to %else the error got fixed.

kevinaboos, 2021-05-14:
> I think I've figured out the problem. It seems to be happening because the code on line 17 is %elif, not %else. When I changed it to %else the error got fixed.

@aghoneim92  Oh, great catch! I completely missed that, you're right.

Much appreciated. Would you like to submit a PR so we can credit you with the fix? If not, I can just give you a shout-out in the commit msg.

kevinaboos, 2021-05-19:
@aghoneim92 friendly ping -- would you like to submit a PR for this? 
Or, @virtuehive, since you initially discovered the issue?

closed 2021-06-01 by kevinaboos

sascha1337, 2022-07-17:
> also planning support for RISC-V within the next year or so. 

what is missing sir ? what are the bl0ckers here

kevinaboos, 2022-07-18:
> > also planning support for RISC-V within the next year or so.
> 
> what is missing sir ? what are the bl0ckers here

Free time 😅

Also, it's looking like we will prioritize aarch64 over risc-v.

PR #392 (2021-06-01): Fix build on MacOS Catalina
Fixes #367

Issue first reported by @virtuehive  and fix determined by @aghoneim92, but since no one was making the PR I took it upon myself.
commit bd408c7e (2021-06-01): Fix compilation on MacOS. Closes #367 (#392)

* Issue first reported by @virtuehive and fixed by @aghoneim92