theseus-os/Theseus #379: Auto-generate Theseus book on each commit

kevinaboos, 2021-05-13:
How to do so: https://github.com/rust-lang/mdBook/blob/master/guide/src/continuous-integration.md

apogeeoak, 2021-06-15:
Hello. I would like to implement this, but I have some questions.

Would it be possible to separate the book into a dedicated repository? A decoupled book makes contributing to the book easier. Orientation is easier as there is less material to orient to. Building is easier and more idiomatic. Testing and the other functionality of `mdBook` are easier to access.

For the actual integration, should it be implemented using Travis, similar to the main project and the linked instructions, or using GitHub Actions as mentioned in the wiki?

kevinaboos, 2021-06-17:
Hi @apogeeoak, thanks for volunteering to help out! Sorry for the delay, I'm on vacation.

For now I'd like to keep the book in the same repository, as it's easier to link against source code or other scripts/artifacts more easily. We may move it to a separate repo in the future.

I think using github actions to build the documentation (both the book and rustdoc) is a good approach. The Travis yaml file that is in the repo is actually not being used, it's an old artifact that can be removed. I don't have much experience with CI tools myself, so I'm open to advice & recommendations here. Let me know how you'd like to proceed and I'll help out where I can. Thanks!

kevinaboos, 2021-06-26:
Closed in ea7447a34d3a20f21d352213a36122f56844e9bb

closed 2021-06-26 by kevinaboos