theseus-os/Theseus #406: "Published academic papers" hyper link in read me is broken

w62, 2021-07-08:
The link https://github.com/theseus-os/Theseus/blob/theseus_main/book/src/papers_presentations.md is broken.

w62, 2021-07-08:
I have found [it](https://theseus-os.github.io/Theseus/book/misc/papers_presentations.html).

kevinaboos, 2021-07-08:
Hi @w62, thanks for the report! I'm glad you found the right link, but can you let me know where you found that dead link initially? I'd like to fix it for others in the future.

kevinaboos, 2021-07-08:
Thanks, I found the links in the README. Appreciate you finding the dead links!

closed 2021-07-08 by kevinaboos

w62, 2021-07-12:
> Hi @w62, thanks for the report! I'm glad you found the right link, but can you let me know where you found that dead link initially? I'd like to fix it for others in the future.

Hi Kevin, 

I just click follow book style documentation, then click Papers and Presentations/Slides at the very bottom.

william

kevinaboos, 2021-07-12:
Thanks, can you try it again now? Everything should be fixed.

w62, 2021-07-13:
> Thanks, can you try it again now? Everything should be fixed.

Yes, it is working fine now. Thanks for resolving the issue.

kevinaboos, 2021-07-13:
Great! Thanks again for bringing it to our attention.

commit 7c48b812 (2021-07-08): Fix links to point to the newly-hosted Theseus Book on GitHub pages.

Closes #406.
commit 618fe9b2 (2021-07-08): Fix links to point to the newly-hosted Theseus Book on GitHub pages.

Closes #406. 7c48b8129a8de095f9a7dd0218ab4afc84bd769c