theseus-os/Theseus #413: Question: How does TheseusOS advoid state spill if a server MUST know some information about a client?

ali-fire-dev, 2021-07-20:
TheseusOS seems great !  May I have a question?
    We know that state spill occurs in Android’s ClipboardService and AlarmManagerService, which may  cause user applications not to work  properly when these service restart after a crash. For example, when ClipboardService restarts, it loses previous data such as some copied texts, and causes the client unable  to paste.
    I wonder that how TheseusOS solves this problem if the service (similar to ClipboardService) in TheseusOS doesn't hold any information (such as the copied texts to paste) about the client. 
    If a ClipboardService doesn't hold any texts from client, how can it works?
    Thank you for your reply！

kevinaboos, 2021-07-22:
Hi @ali-fire-dev, thanks for the question! Recently we've moved away from a focus on state spill, but I'm happy to answer this question.

For this specific type of spill, there's no clean/easy way to avoid it. There are some implementation tricks we can leverage, such as externalizing server-side state such that a crashed service could restore its checkpointed state upon restart, or using server-side weak references to client-owned state such that the server doesn't actually own any state. In general, some forms of spill are unavoidable (especially if you wish to enable certain kinds of functionality), so here it'd be more about avoiding/mitigating the adverse effects of state spill with one of the aforementioned implementation tricks.

ali-fire-dev, 2021-07-26:
> Hi @ali-fire-dev, thanks for the question! Recently we've moved away from a focus on state spill, but I'm happy to answer this question.
>
> For this specific type of spill, there's no clean/easy way [...]

I see. Thank you for your reply!

closed 2021-07-26 by ali-fire-dev