theseus-os/Theseus #414: Safety of slabmalloc's deallocate()

Qwaz, 2021-07-21:
Hello, I was testing a static analysis tool that validates unsafe usage with Theseus and would like to share the finding from it.

https://github.com/theseus-os/Theseus/blob/2bc18d736ac0a910cf747e45bde8b1f91c700fb1/kernel/slabmalloc/src/sc.rs#L311-L327

https://github.com/theseus-os/Theseus/blob/2bc18d736ac0a910cf747e45bde8b1f91c700fb1/kernel/slabmalloc/src/zone.rs#L188-L194

`slabmalloc` and `slabmalloc_unsafe` libraries expose `SCAllocator::deallocate` and `ZoneAllocator::deallocate` as public safe APIs. However, they transmute the argument pointer to a slab page without check. Thus, they have a safety invariant on the caller side that affects memory safety and should probably be defined as unsafe functions.

Their impact is minimal, since Theseus is the only user of the libraries, and as far as I can tell all use in Theseus (multiple_heaps) seem correct. Nevertheless, I think defining them as unsafe might better express the API safety and conform to Rust's safety rules.

I wonder if the maintainers are willing to accept a PR that updates the signature of those functions. The update can be limited to `slabmalloc` and `slabmalloc_unsafe`, or it can additionally include `slabmalloc_safe` for symmetricity.

Thanks!

kevinaboos, 2021-07-22:
Hi @Qwaz, thanks for bringing this to our attention! Yes, we're totally open to PRs and changes. Some of those libraries are just for testing/benchmarks and aren't actually used, and can probably be removed at this point. But in general, it seems that you're correct, they shouldn't be public. I presume they were made public for easy testing and benchmarking through invoking them from other crates. 

I think @Ramla-I may be interested in your findings too. 

I think the entire heap design can be re-done and made wholly safe, but we haven't really had a chance to do that yet. We're basically just using thin wrappers around an existing unsafe slabmalloc crate.

Ramla-I, 2021-07-23:
Thanks for pointing this out @Qwaz! I agree, those functions should be marked unsafe, and I think `SCAllocator::deallocate` can actually be made into a private function.

PR #424 (2021-08-24): Adjust safety of deallocate() functions
This PR adjusts the declared safety of `deallocate()` functions. It also made `SCAllocator` private to its corresponding crate since the allocation/deallocation functionality is exposed through `ZoneAllocator` APIs. Fixes #414.
closed 2021-08-24 by kevinaboos