theseus-os/Theseus #425: Missing/invalid symbol expected from assembly code "ap_start_realmode"

ZetaNumbers, 2021-08-13:
Invoked command:
`make run`

Stdout:
```
=================== BUILDING ALL CRATES ===================
	 TARGET: "x86_64-theseus"
	 KERNEL_PREFIX: "k#"
	 APP_PREFIX: "a#"
	 THESEUS_CONFIG (before build.rs script): ""
RUST_TARGET_PATH="/home/zeta0/Projects/Theseus/cfg" RUSTFLAGS="--emit=obj -C debuginfo=2 -C code-model=large -C relocation-model=static -D unused-must-use -Z merge-functions=disabled -Z share-generics=no" cargo build --release -Z unstable-options -Z build-std=core,alloc -Z build-std-features=compiler-builtins-mem  --all --target x86_64-theseus
ld -n -T /home/zeta0/Projects/Theseus/kernel/nano_core/src/boot/arch_x86_64/linker_higher_half.ld -o /home/zeta0/Projects/Theseus/build/nano_core/nano_core-x86_64.bin  /home/zeta0/Projects/Theseus/build/nano_core/boot/x86_64/ap_boot.o  /home/zeta0/Projects/Theseus/build/nano_core/boot/x86_64/ap_realmode.o  /home/zeta0/Projects/Theseus/build/nano_core/boot/x86_64/boot.o  /home/zeta0/Projects/Theseus/build/nano_core/boot/x86_64/common.o  /home/zeta0/Projects/Theseus/build/nano_core/boot/x86_64/defines.o  /home/zeta0/Projects/Theseus/build/nano_core/boot/x86_64/multiboot_header.o /home/zeta0/Projects/Theseus/target/x86_64-theseus/release/libnano_core.a
'/home/zeta0/Projects/Theseus/cfg/x86_64-theseus.json' -> '/home/zeta0/Projects/Theseus/build/deps/x86_64-theseus.json'
make grub
make[1]: вход в каталог «/home/zeta0/Projects/Theseus»
make[1]: выход из каталога «/home/zeta0/Projects/Theseus»
qemu-system-x86_64 -cdrom /home/zeta0/Projects/Theseus/build/theseus-x86_64.iso -boot d -no-reboot -no-shutdown -s -serial stdio  -m 512M  -smp 4 -net none -cpu Broadwell
[I] kernel/nano_core/src/lib.rs:106: Logger initialized.

nano_core_start(): initialized logger.

exceptions_early(): double_fault_stack_top_unusable: Some(v0xFFFFFFFF80612000)

exceptions_early(): Created TSS: TaskStateSegment { reserved_1: 0, privilege_stack_table: [0x0, 0x0, 0x0], reserved_2: 0, interrupt_stack_table: [0xffffffff80612000, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0], reserved_3: 0, reserved_4: 0, iomap_base: 0 }

nano_core_start(): initialized early IDT with exception handlers.

nano_core_start(): booted via multiboot2 with info at 0xFFFFFFFF816D58A0.

[D] kernel/memory/src/lib.rs:156: bootloader info memory: p0x16D58A0 to p0x16D93A0, bootloader modules: p0x4D4000 to p0x16D5898

[D] kernel/memory/src/lib.rs:159: kernel_phys_start: p0x100000, kernel_phys_end: p0x612000 kernel_virt_end = v0xffffffff80612000

[D] kernel/memory/src/lib.rs:207: Initialized new frame allocator!

[D] kernel/frame_allocator/src/lib.rs:922: ----------------- FREE GENERAL FRAMES ---------------

[D] kernel/frame_allocator/src/lib.rs:923: 	 Chunk { typ: Free, frames: Frame(p0x16DA000)..=Frame(p0x1FFDF000) }

[D] kernel/frame_allocator/src/lib.rs:924: -----------------------------------------------------

[D] kernel/frame_allocator/src/lib.rs:925: ----------------- FREE RESERVED FRAMES --------------

[D] kernel/frame_allocator/src/lib.rs:926: 	 Chunk { typ: Reserved, frames: Frame(p0x0)..=Frame(p0xFF000) }

[D] kernel/frame_allocator/src/lib.rs:926: 	 Chunk { typ: Reserved, frames: Frame(p0x100000)..=Frame(p0x16D9000) }

[D] kernel/frame_allocator/src/lib.rs:926: 	 Chunk { typ: Reserved, frames: Frame(p0x1FFE0000)..=Frame(p0x1FFFF000) }

[D] kernel/frame_allocator/src/lib.rs:926: 	 Chunk { typ: Reserved, frames: Frame(p0xFFFC0000)..=Frame(p0xFFFFF000) }

[D] kernel/frame_allocator/src/lib.rs:927: -----------------------------------------------------

[D] kernel/frame_allocator/src/lib.rs:928: ------------------ RESERVED REGIONS -----------------

[D] kernel/frame_allocator/src/lib.rs:929: 	 Chunk { typ: Reserved, frames: Frame(p0x0)..=Frame(p0xFF000) }

[D] kernel/frame_allocator/src/lib.rs:929: 	 Chunk { typ: Reserved, frames: Frame(p0x100000)..=Frame(p0x16D9000) }

[D] kernel/frame_allocator/src/lib.rs:929: 	 Chunk { typ: Reserved, frames: Frame(p0x1FFE0000)..=Frame(p0x1FFFF000) }

[D] kernel/frame_allocator/src/lib.rs:929: 	 Chunk { typ: Reserved, frames: Frame(p0xFFFC0000)..=Frame(p0xFFFFF000) }

[D] kernel/frame_allocator/src/lib.rs:930: -----------------------------------------------------

[D] kernel/memory/src/lib.rs:211: Initialized new page allocator!

[D] kernel/page_allocator/src/lib.rs:682: --------------- FREE PAGES LIST ---------------

[D] kernel/page_allocator/src/lib.rs:684: Chunk { pages: Page(v0x0)..=Page(v0x612000) }
[... 178 lines cut ...]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.tsc-deadline [bit 24]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.avx [bit 28]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.f16c [bit 29]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.avx2 [bit 5]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
```

System info:
```yaml
---
host:
  name: Arch Linux
  os version: rolling
  kernel version: 5.13.9-arch1-1
cpu:
  - Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz
  - Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz
  - Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz
  - Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz
```

kevinaboos, 2021-08-13:
Hi @ZetaNumbers, thanks for filing this issue! 

I'm unable to reproduce this on a clean copy of Theseus, unfortunately. I assume you encountered this error when checking out the latest version of this repository and doing a new clean build? If not, try that out and let me know. 

My best guess is that something may be different with your assembler, since it's failing to find an expected constant that's defined in assembly. Perhaps this has something to do with the differences between building on Arch vs. a Debian-based distro (?). If so, we'd need to pin down the correct version of `nasm` and/or other related tools.

Even though I can't reproduce it, I'd like to help figure this out. Can you provide more details about your version of each tool used in Theseus? Let's start with `nasm`, but we might as well check the versions for each of the dependencies specified in the README: `make gcc nasm pkg-config grub-pc-bin mtools xorriso qemu qemu-kvm`.

kevinaboos, 2021-08-13:
Additionally, I'm quite interested how this could actually occur after a successful build, because I added `extern` symbol declarations to the Rust-level code to prevent this very problem -- linking should fail if those constants are missing from the asm level. See here: https://github.com/theseus-os/Theseus/blob/fbe80b056c75b23e4b51cf0bbd47cf17f05da671/kernel/nano_core/src/lib.rs#L239-L249

Hopefully we can get to the bottom of this, thanks!

kevinaboos, 2021-08-13:
Closing this based on #426. 
Feel free to re-open if this issue persists.

ZetaNumbers, 2021-08-13:
> Closing this based on #426.
> Feel free to re-open if this issue persists.

Github doesn't allow me to reopen this issue 🙁

ZetaNumbers, 2021-08-13:
I have updated issue's info (top comment).

So about the symbols. It's not that.
```
$ readelf -sW ./build/nano_core/nano_core-x86_64.bin | grep ap_start_realmode
  8135: 0000000000100102     0 NOTYPE  GLOBAL DEFAULT    1 ap_start_realmode
  8461: 0000000000100376     0 NOTYPE  GLOBAL DEFAULT    1 ap_start_realmode_end
```

ZetaNumbers, 2021-08-13:
So i tried to look where it comes from and i stumbled onto this statement:

```rust
// kernel/mod_mgmt/src/parse_nano_core.rs:104
let parse_result = match nano_core_file_path.extension() {
  Some("sym") => parse_nano_core_symbol_file(&nano_core_crate_ref, &text_pages, &rodata_pages, &data_pages),
  Some("bin") => parse_nano_core_binary(&nano_core_crate_ref, &text_pages, &rodata_pages, &data_pages),
  _ => Err("nano_core object file had unexpected file extension. Expected \".bin\" or \".sym\""),
};
```

I am not familiar with this repository's code, but i think it is strange that `nano_core_file_path` is a symbol file `.sym` and not the `.bin` file as mentioned above. I learnt it from my dirty debug attempts. I may assure you `parse_result` contains no init symbols, where `ap_start_realmode` would be.

kevinaboos, 2021-08-13:
> Github doesn't allow me to reopen this issue  @kevinaboos

No problem, I'll re-open it.

ZetaNumbers, 2021-08-13:
I will say problem arises only if `make iso` have been done on archlinux. Docker container does it fine, to be `make orun` it on archlinux.

Maybe we could add archlinux docker container? I can do that, but the bug will stay.

kevinaboos, 2021-08-13:
Thanks for the extra details, I'd rather fix the issue on Arch instead of creating a separate docker image, since that's hard to maintain. 

Those symbols seem correct in the `nano_core-x86_64.bin` file you posted. The `.sym` file is an optimization to improve parsing time, that's intentional. 

Perhaps the line in the Makefile that generates the `.sym` file is somehow not working on Arch Linux? (That part is L250-252 of the Makefile.)  Maybe something is different with the version of `readelf` or `sed` on Arch, I'm not sure since I've never used Arch. 

Can you confirm that those symbols exist in the .sym file? Try this:
```sh
grep ap_start_realmode build/grub-isofiles/modules/k#nano_core.sym
```
That should output something like:
```
  8716: 00000000001001c7     0 NOTYPE  GLOBAL DEFAULT    1 ap_start_realmode
  9049: 000000000010043b     0 NOTYPE  GLOBAL DEFAULT    1 ap_start_realmode_end
```

ZetaNumbers, 2021-08-13:
> Can you confirm that those symbols exist in the .sym file?

```
$ grep ap_start_realmode 'build/grub-isofiles/modules/k#nano_core.sym'
  8265: 0000000000100102     0 NOTYPE  GLOBAL DEFAULT    1 ap_start_realmode
  8591: 0000000000100376     0 NOTYPE  GLOBAL DEFAULT    1 ap_start_realmode_end
```

ZetaNumbers, 2021-08-13:
I think i found the issue
<details>
<summary>
It's in russian lol
</summary>

```
Имеется 23 заголовка раздела, начиная со смещения 0x1fa8b40:

Заголовки разделов:
  [Нм] Имя               Тип             Адрес            Смещ   Разм   ES Флг Лк Инф Al
  [ 0]                   NULL            0000000000000000 000000 000000 00      0   0  0
  [ 1] .init             PROGBITS        0000000000100000 000120 000526 00  AX  0   0  8
  [ 2] .text             PROGBITS        ffffffff80101000 001000 213caa 00  AX  0   0 16
  [ 3] .rodata           PROGBITS        ffffffff80315000 215000 07296c 00   A  0   0 16
  [ 4] .eh_frame         X86_64_UNWIND   ffffffff80388000 288000 065f64 00   A  0   0  8
  [ 5] .gcc_except_table PROGBITS        ffffffff803ee000 2ee000 00bb2c 00   A  0   0  4
  [ 6] .data             PROGBITS        ffffffff803fa000 2fa000 003040 00  WA  0   0  8
  [ 7] .bss              NOBITS          ffffffff803fe000 2fd040 006a08 00  WA  0   0 64
  [ 8] .page_table       NOBITS          ffffffff80405000 2fd040 001000 00  WA  0   0 4096
  [ 9] .stack            NOBITS          ffffffff80800000 2fe000 012000 00  WA  0   0 4096
  [10] .got              NOBITS          ffffffff80812000 2fe000 000000 00  WA  0   0  8
  [11] .debug_loc        PROGBITS        0000000000000000 2fd040 68e81b 00      0   0  1
  [12] .debug_abbrev     PROGBITS        0000000000000000 98b85b 01b230 00      0   0  1
  [13] .debug_info       PROGBITS        0000000000000000 9a6a8b 7728a8 00      0   0  1
  [14] .debug_aranges    PROGBITS        0000000000000000 1119333 026290 00      0   0  1
  [15] .debug_ranges     PROGBITS        0000000000000000 113f5c3 2671b0 00      0   0  1
  [16] .debug_str        PROGBITS        0000000000000000 13a6773 4d61e2 01  MS  0   0  1
  [17] .debug_pubnames   PROGBITS        0000000000000000 187c955 1e7ddb 00      0   0  1
  [18] .debug_pubtypes   PROGBITS        0000000000000000 1a64730 2cb336 00      0   0  1
  [19] .debug_line       PROGBITS        0000000000000000 1d2fa66 17ed6e 00      0   0  1
  [20] .symtab           SYMTAB          0000000000000000 1eae7d8 03ae48 18     21 5382  8
  [21] .strtab           STRTAB          0000000000000000 1ee9620 0bf437 00      0   0  1
  [22] .shstrtab         STRTAB          0000000000000000 1fa8a57 0000e7 00      0   0  1
Обозначения флагов:
  W (запись), A (назнач), X (исполняемый), M (слияние), S (строки),
  I (инфо), L (порядок ссылок), O (требуется дополнительная работа ОС),
  G (группа), T (TLS), C (сжат), x (неизвестно), o (специфич. для ОС),
  E (исключён),
  l (большой), p (processor specific)

Таблица символов «.symtab» содержит 10051 элемент:
   Чис:    Знач           Разм Тип     Связ   Vis      Индекс имени
   161: ffffffff80101170    20 FUNC    LOCAL  DEFAULT    2 _ZN43_$LT$logger..Logger$u20$as$u20$log..Log$GT$7enabled17h633c794ce56cf98dE
   162: ffffffff80101190    25 FUNC    LOCAL  DEFAULT    2 _ZN44_$LT$$RF$T$u20$as$u20$core..fmt..Display$GT$3fmt17h934678693d4d9a9aE
   163: ffffffff801011b0    22 FUNC    LOCAL  DEFAULT    2 _ZN4core3ptr101drop_in_place$LT$irq_safety..mutex_irqsafe..MutexIrqSafeGuard$LT$memory..MemoryManagementInfo$GT$$GT$17hc408c975bcdf3f56E
   164: ffffffff801011d0   251 FUNC    LOCAL  DEFAULT    2 _ZN4core3ptr113drop_in_place$LT$hashbrown..map..HashMap$LT$usize$C$alloc..sync..Arc$LT$crate_metadata..LoadedSection$GT$$GT$$GT$17h589fa0a20ebb1b3eE
...
```

</details>

kevinaboos, 2021-08-13:
Oh, that's interesting! Yeah, I could tell that something with the parsing was incorrect, since this log line showed a failure to parse anything useful from the symbol file.
```
[I] kernel/mod_mgmt/src/parse_nano_core.rs:131: Finished parsing nano_core crate, 0 new symbols.
```

I hadn't considered localization differences when parsing the symbol file, that's clearly the problem. Thanks for identifying this!

kevinaboos, 2021-08-13:
I had intended to use a binary format to (de)serialize the nano_core symbols for faster text-less parsing, but just hadn't gotten around to it yet. That would clearly solve this problem, but it's a large undertaking. I think a good interim solution would be to pin the locale of various Makefile commands to be en_US. 

If you have another solution in mind, I'm open to it. I'm also open to accepting a PR to fix this, since I may not be able to address it for a while.

ZetaNumbers, 2021-08-13:
I would recommend using libraries like [object](https://crates.io/crates/object) or [goblin](https://crates.io/crates/goblin) for reading the binaries.

kevinaboos, 2021-08-13:
> I would recommend using libraries like [object](https://crates.io/crates/object) or [goblin](https://crates.io/crates/goblin) for reading the binaries.

Yes, we do that. That's the other option that was previously used, see `parse_nano_core_binary()`. However, it's extremely slow, so we deprecated that by default.

ZetaNumbers, 2021-08-13:
I am familiar with these crates. I would like to try to develop a fix but i have a work to do at this moment, tho i should be free in the near future. :)

kevinaboos, 2021-08-13:
Same, I've used them before in addition to `xmas-elf`. 

Anyway, thanks for bringing this to my attention. Obviously, parsing english text isn't a great solution here, but it's also a low-priority problem at this point. Parsing the actual binary file is prohibitively expensive during bootstrap, so I wouldn't recommend trying that solution again. Serializing the symbol file into a non-human-readable binary format at build time and then deserializing it into memory at bootstrap time is my ideal solution, but as I said before I haven't yet gotten around to implementing that. 

For now, I'll probably implement a quick fix to avoid searching for specific English phrases. :+1:

kevinaboos, 2021-08-13:
@ZetaNumbers  I've pushed a quick fix for this in #430. Can you let me know if that works for you?

closed 2021-08-16 by kevinaboos

ZetaNumbers, 2021-08-18:
Coming back to this issue, I think we could parse `rust-readobj` from [rust-embedded/cargo-binutils](https://github.com/rust-embedded/cargo-binutils/) instead of `readelf`. It is easy to install and does not depend on your locale.

However i think #431 is preferable, so i think we might hold on that.

ZetaNumbers, 2021-09-01:
Quick fix: `LANG="en_US.UTF-8" make run host=yes`

PR #430 (2021-08-16): Parsing the nano_core symbol file is now locale-independent
Should fix #425.
commit 21da16cf (2021-08-16): Parsing the nano_core symbol file is now locale-independent. Closes #425. (#430)
commit 982ddd4e (2021-08-16): Parsing the nano_core symbol file is now locale-independent. Closes #425. (#430) 21da16cffe3e516e2bc710c66d8f7584da80c186