theseus-os/Theseus #426: `tools/demangle_readelf_file` requires `GLIBC_2.33`

ZetaNumbers, 2021-08-13:
Hi. I tried to use docker method, but got an error where `demangle_readelf_file` requires `GLIBC_2.33` from `libc.so.6`. The problem is that Ubuntu 18.04 only supports `GLIBC_2.27`.

kevinaboos, 2021-08-13:
Hi @ZetaNumbers, thanks for the issue report! 

That's odd, I'm running both Ubuntu 18.04 and 20.04 and I don't have any issues, but I also don't use the docker image. My version of glibc is 2.31, which works without problems. 

I'll take a look at your PR as well. Thanks!

ZetaNumbers, 2021-08-13:
https://github.com/theseus-os/Theseus/pull/427#issuecomment-898678248

kevinaboos, 2021-08-13:
haha, thanks, and no worries. There's a semi-related issue (#421) introduced by one of my latest changes, but it's not docker-specific. That will be addressed shortly. 

Regarding the sharing of target and build directories with the container and host, that behavior is intentional. It allows you to keep all build dependencies within the container, but then exit the container after the build completes and run the built version of Theseus on your host (e.g., using KVM for acceleration). This functionality was necessary for easily reproducing our benchmarks for our prior academic publications.

closed 2021-08-13 by kevinaboos