theseus-os/Theseus #428: `make clean` doesn't clean

ZetaNumbers, 2021-08-13:
```make
### Removes all built source files
clean:
	cargo clean
	@rm -rf $(BUILD_DIR)
```
actually ignores several target dirs, for example `tools/demangle_readelf_file/target`

kevinaboos, 2021-08-13:
That is true, but is it necessary to clean those target directories? They're just tools that are always executed with `cargo run`, meaning they will only be rebuilt if necessary. 

Generally, these tools only need to be built once after building Theseus for the first time (unless you're modifying the tools themselves). If we clean those tools' target dirs on every clean build, then we're imposing unnecessary build delays for things that are unchanged and not a part of the actual source code. 

Do you think this was the source of the problems you experienced in #427 or other issues?

ZetaNumbers, 2021-08-13:
> but is it necessary to clean those target directories?

`cargo clean` cleans also the build dependecies, so i guess it is. I think many would believe `make clean` would erase *every* build artifact. If you need to clean specifically the kernel (or how is it even called) arifacts, you should specify it.

> Do you think this was the source of the problems you experienced in #427 or other issues?

May be

kevinaboos, 2021-08-13:
Thanks for the feedback. 

> I think many would believe make clean would erase every build artifact

Perhaps, but I'd argue those aren't really build artifacts, they're tools used in the build. The default `clean` target should not remove those by default, in my personal opinion. Similarly, one wouldn't expect other build tools like gcc, make, nasm, etc to be deleted and then re-built from scratch on every build invocation. 

If this is a demonstrable problem, then we can add another target to clean the tools directories. In fact, it's standard Makefile practice to have `clean` only remove compiled source code artifacts, leaving other targets to remove various items, e.g., `make distclean`, `make maintainer-clean`, etc. 

As of now, I don't see it being problematic, even in the docker container scenario of sharing files between the host and container environment. Please do correct me if you determine otherwise.

ZetaNumbers, 2021-08-13:
Well then how would i remove `tools/demangle_readelf_file/target`? This is where executable required `GLIBC_2.33` in my other issue. Btw this problem probably can occur with the other tools too. 

> Perhaps, but I'd argue those aren't really build artifacts

I am just saying majority of the people would have expected `make clean` to remove any trace of compilation. I am just proposing to clarify these things, like add `clean_tools` and `clean_nontools` and clarify that `clean: clean_tools clean_nontools` (or maybe remove `clean` if you would want that).

kevinaboos, 2021-08-13:
> Well then how would i remove `tools/demangle_readelf_file/target`? This is where executable required `GLIBC_2.33` in my other issue. Btw this problem probably can occur with the other tools too.

You can run `cargo clean` in that directory. I agree this would occur in other tools, but I have yet to encounter this problem.

My point is that there is generally no need to clean these tools, since they will never change across builds. If I published them as separate binaries (where you'd get them via `cargo install <tool>`) instead of packaging them in the Theseus repo, then they also wouldn't be cleaned/deleted when running `make clean`; that's what the expected behavior is and should be. 

> I am just saying majority of the people would have expected `make clean` to remove any trace of compilation. I am just proposing to clarify these things, like add `clean_tools` and `clean_nontools` [...]

I disagree, that's not necessarily what `clean` does. The [Linux kernel does not abide by that rule](https://www.kernel.org/doc/makehelp.txt) either, for what it's worth.

In any case, I am happy to accept a new clean target called `clean-tools`, which will run `cargo clean` on every subdirectory in `tools/` that contains a `Cargo.toml`. However, the existing `clean` target should remain unchanged.

closed 2021-08-14 by ZetaNumbers