theseus-os/Theseus #444: Rustc plugin/custom rustc as a safety invariant. Discussion

ZetaNumbers, 2021-09-16:
I saw WIP since 2018 plugin there. I am trying to investigate a possibility of a custom rustc (or plugin). As i understand it, this tool should forbid unsafe code for any except kernel crates. The issue would be the generated code (`#![forbid(unsafe_code)]` allows unsafe code to be generated by also `#![forbid(unsafe_code)]` crate). I am trying to get a crate id or something from a span of an unsafe block, to validate it.

I would like to know if there were perhaps any roadblocks i am not hitting yet?

kevinaboos, 2021-09-18:
Hi @ZetaNumbers, sorry but I'm not exactly sure what you're asking here. It's been a few years since I experimented with compiler plugins, and some of my academic colleagues had helped in the past with a variety of approaches. I'm not very knowledgeable in that area.

SamuraiCrow, 2021-12-07:
I think that what is trying to be described here is a way to avoid having to arbitrarily audit code to look for unsafes and other undesirable characteristics that might be inappropriate for the safety of Theseus code.  The ability to flag unsafes as unacceptable or requiring a code audit on those that don't contain that extension could help automate code review exercises.

Am I missing the point here?

Nutomic, 2022-05-27:
For detecting unsafe usage, [cargo-geiger](https://github.com/rust-secure-code/cargo-geiger) seems like the correct tool.

ZetaNumbers, 2022-06-03:
> For detecting unsafe usage, [cargo-geiger](https://github.com/rust-secure-code/cargo-geiger) seems like the correct tool.

I am afraid it does not detect unsafe generated from macros, nor within a macro definition. My idea for unsafe detection was to look at the crate's code after macro expansion and allow `unsafe` keyword only from the "kernel" crates (via span). Sadly, i couldn't figure out at a time how to determine which crate (from some crate whitelist) a token is from.