theseus-os/Theseus #447: GUI / Desktop Environment

aghoneim92, 2021-09-30:
How would one go about developing a GUI for Theseus? I am willing to tinker and investigate, but would like to know if you have any pointers (pun-intended) for me

kevinaboos, 2021-10-04:
Hi @aghoneim92 , thanks for your interest! There is a very basic software compositor, window manager, and support for displayable items already; in fact, the shell window that is shown by default when Theseus first boots up is a graphical window itself.

The graphics subsystem definitely needs a re-work since it was designed in a hurry with little effort. That being said, it is well-documented, so you can check out the source code docs for window-related crates. Another good place to get started would be the [Theseus Book chapter on graphical display](https://theseus-os.github.io/Theseus/book/subsystems/display/display.html) support.
There are many directions that graphics-related project could go, so if you're interested in a particular aspect of the graphics subsystem that isn't covered in the book chapter, please let me know.

closed 2021-11-29 by kevinaboos

CohenAriel, 2022-07-25:
What about using an external WM? For example https://github.com/sminez/penrose or one made in C like gnome (though it wouldn't be safe).