theseus-os/Theseus #507: Could not initialize SDL(No available video device) - exiting Makefile:771: recipe for target 'run' failed

nbro, 2022-04-21:
I was trying to build Rust on a docker container, as explained in the README file, but I am getting the error:

> Could not initialize SDL(No available video device) - exiting 
> Makefile:771: recipe for target 'run' failed

Here's the full traceback

```
info: downloading component 'rust-src'
info: installing component 'rust-src'

=================== BUILDING ALL CRATES ===================
	 TARGET: "x86_64-theseus"
	 KERNEL_PREFIX: "k#"
	 APP_PREFIX: "a#"
	 THESEUS_CONFIG (before build.rs script): ""
RUST_TARGET_PATH="/Theseus/cfg" RUSTFLAGS="--emit=obj -C debuginfo=2 -C code-model=large -C relocation-model=static -D unused-must-use -Z merge-functions=disabled -Z share-generics=no" cargo build  --release -Z unstable-options -Z build-std=core,alloc -Z build-std-features=compiler-builtins-mem  --all --target x86_64-theseus
    Updating git repository `https://github.com/theseus-os/getopts`
    Updating git repository `https://github.com/theseus-os/qp-trie-rs`
    Updating git repository `https://github.com/m-labs/smoltcp`
    Updating git repository `https://github.com/theseus-os/spin-rs`
    Updating git repository `https://github.com/theseus-os/volatile`
    Updating git repository `https://github.com/theseus-os/irq_safety`
    Updating crates.io index
    Updating git repository `https://github.com/theseus-os/xmas-elf.git`
    Updating git repository `https://github.com/theseus-os/owning-ref-rs`
    Updating git repository `https://github.com/rafalh/rust-fatfs`
    Updating git repository `https://github.com/theseus-os/rbtree-rs.git`
    Updating git repository `https://github.com/theseus-os/stable_deref_trait.git`
    Updating git repository `https://github.com/rust-lang/stdsimd`
    Updating git repository `https://github.com/bytecodealliance/wasi`
    Updating git submodule `https://github.com/WebAssembly/WASI`
    Updating git repository `https://github.com/theseus-os/zero.git`
  Downloaded bit_field v0.10.1
  Downloaded byteorder v1.4.3
  Downloaded cfg-if v1.0.0
  Downloaded const_format_proc_macros v0.2.22
  Downloaded cc v1.0.71
  Downloaded hashbrown v0.11.2
  Downloaded convert_case v0.4.0
  Downloaded by_address v1.0.4
  Downloaded getrandom v0.2.4
  Downloaded scopeguard v1.1.0
  Downloaded generic-array v0.12.0
  Downloaded digest v0.8.0
  Downloaded lock_api v0.4.2
  Downloaded itertools v0.7.11
  Downloaded memory_units v0.3.0
  Downloaded log v0.4.8
  Downloaded paste v1.0.5
  Downloaded memoffset v0.5.6
  Downloaded once_cell v1.9.0
  Downloaded rustc-demangle v0.1.19
  Downloaded rustc_version v0.2.3
  Downloaded num_enum v0.5.2
  Downloaded memchr v2.4.1
  Downloaded rand_chacha v0.1.0
  Downloaded rand_pcg v0.1.1
  Downloaded raw-cpuid v7.0.3
  Downloaded scroll v0.9.2
  Downloaded stable_deref_trait v1.1.1
  Downloaded void v1.0.2
  Downloaded unreachable v1.0.0
  Downloaded semver v0.9.0
  Downloaded volatile v0.4.4
  Downloaded wasmi-validation v0.4.1
  Downloaded quote v1.0.7
  Downloaded zerocopy v0.5.0
  Downloaded wasmi v0.9.1
  Downloaded ahash v0.7.6
[... 487 lines cut ...]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.fma [bit 12]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.pcid [bit 17]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.tsc-deadline [bit 24]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.avx [bit 28]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.f16c [bit 29]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.rdrand [bit 30]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.avx2 [bit 5]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
Could not initialize SDL(No available video device) - exiting
Makefile:771: recipe for target 'run' failed
make: *** [run] Error 1
```

There are several related posts on the web about this error. See, for example, [here](https://stackoverflow.com/q/22967925/3924118). However, given that I am not really familiar with Theseus (I'm trying it for the first time), I don't want to spend like 10 days trying to figure out the problem, while maybe someone can solve it in 5 minutes.

My host OS is Mac OS Monterey (12.3.1).

kevinaboos, 2022-04-21:
Hi @nbro, thanks for the question! I believe the issue comes from being unable to run qemu in a docker container, though admittedly it's been awhile since I've tried to do so.

If you exit the container after building Theseus and run qemu on your host machine, then it ought to work. This is what our docker scripts and documentation suggests doing for KVM support, and on a Mac it may be required for qemu even without KVM.

kevinaboos, 2022-04-21:
Oh wait, it looks like I missed the key issue. The good thing is that this doesn't have anything to do with Theseus specifically, it's about QEMU being unable to find an X window server on macOS.

I think the solution might be accomplished with one or two settings, but unfortunately I don't have access to a macOS machine at the moment and thus cannot test it myself.
* Instal an X windows server on macOS and configuring your system to use it.
   * If you take a look at the instructions for building on WSL, you'll see that we have to install something like `Xming` and the set the corresponding `DISPLAY` environment variable, e.g., `export DISPLAY=...`.
   * On macOS, I believe the corresponding program is called XQuartz.
* If that's not enough, perhaps you need to run the docker container in a privileged mode such that it has access to macOS's X window server.

Hopefully that will get things working for you!

kevinaboos, 2022-04-27:
Closing due to issue being addressed. Feel free to re-open if still problematic.

nbro, 2022-04-27:
@kevinaboos I didn't have time to look at your suggestions and try them so far. I would leave the issue open until we're sure that it's really solved.

kevinaboos, 2022-04-27:
Sure, no problem. Let us know if it works for you.

closed 2023-01-23 by kevinaboos