theseus-os/Theseus #526: Fix cargo clippy warnings

xa888s, 2022-06-10:
Running `cargo clippy` yields over ~6500 lines of warnings in it's output. Most of these are pretty simple fixes with some false positives here and there (we could just use #[allow] directives in these cases). Fixing these would help make the code a little more idiomatic as well as reduce some visual complexity with no change in runtime behaviour. Somewhat related to #35 in regards to general code base maintenance.

kevinaboos, 2022-06-10:
Hi Brandon, thanks for bringing this up! Yep, this has definitely been on our to-do list for quite a while. I do use clippy in some cases but it's not yet integrated as part of the CI workflow, nor has it been used across the entire repo as a whole.

Kindly feel free to submit a series of PRs that address the linter issues, preferably on a per-subsystem basis. Try to keep the PRs smaller in scope so there aren't 6000+ changes to review in a single PR.

It'd also be great to have GitHub actions workflow that runs it on changed code, if you're so inclined.

kevinaboos, 2022-06-13:
After some thought, I think the best way to do this is to add a new CI action that runs only a subset of approved clippy actions on newly-added (or modified) code only. This way we don't have to worry about retroactively applying 6000+ changes to the entire code base.

Also, this issue can serve as the tracking issue for integrating clippy lints into Theseus at large. We should add a long list of checkboxes to your initial post above, with each checkbox being a subset of available clippy lints. Then, when we have fixed all code such that it passes a given subset of clippy lints, we can check off those lints from the list. The first few can be the ones about `unwrap_or` vs `unwrap_or_else`, thanks to your PR #527. 

We can take a similar approach to `rustfmt`, but I think that's a bit more complex and significantly lower priority. Let's definitely separate them though.

kevinaboos, 2022-06-15:
@xa888s I added the full list of Clippy lints to your original post. Once you undo the rustfmt changes to #527 and we merge it in, you can check those boxes.

kevinaboos, 2022-07-14:
@nramos0 since you're currently doing some great work on this issue, would you be interested in adding a GitHub actions pass that runs the above checked-off clippy lints on changed files in PRs? That way we'd avoid re-introducing code that violates the lints we've already marked as completed.

nramos0, 2022-07-17:
> @nramos0 since you're currently doing some great work on this issue, would you be interested in adding a GitHub actions pass that runs the above checked-off clippy lints on changed files in PRs? That way we'd avoid re-introducing code that violates the lints we've already marked as completed.

Sure, I'll give it a shot.

kevinaboos, 2022-07-21:
> Sure, I'll give it a shot.

Fantastic! Let me know if we can help.

nramos0, 2022-07-22:
When testing the github action for PR #574 I think I may have found another file that was not reached by the clippy build/check: [applications/bm/src/lib.rs](https://github.com/nramos0/Theseus/pull/1/commits/57d565d66a5a12d18434781d033a7cee8873c7c7#diff-0f92a40677145a6ed4e5bd0b586ceb2c56a8f62c354acf551e789e6fe214f010). It passed the check [here](https://github.com/nramos0/Theseus/runs/7474406165?check_suite_focus=true), but the check correctly failed when changing another file [here](https://github.com/nramos0/Theseus/pull/1/commits/8b9feeb3eb6759d3d1498323e089b925d6f981ce).

dcampbell24, 2022-09-10:
I ran the below on the main branch and didn't get any warnings about borrowed_box. Are the not checked off items inclusive of items that have never been tested? Can we update the list to check off all the items that do not exist (and add them to be checked against)?
 
$ cargo clean
$ cargo clippy -Z build-std=core,alloc -Z build-std-features=compiler-builtins-mem --target cfg/x86_64-theseus.json --workspace -- -A clippy::all -D clippy::borrowed_box

I would be willing to run all the tests as above to confirm the issues actually exist.
 
Curiously however, I did get the one warning:

warning: unused import: `frame_allocator::AllocatedFrames`
  --> kernel/page_table_entry/src/lib.rs:20:5
   |
20 | use frame_allocator::AllocatedFrames;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `page_table_entry` (lib) generated 1 warning

tsoutsman, 2022-09-12:
#636 found all the already passing complexity lints.

kevinaboos, 2022-09-13:
Thanks @dcampbell24! Does this include all of the crates, even optional ones? (specified using `make full` or by passing  `--all-features` to cargo)? 
 I'm traveling at the moment and can't check myself.

kevinaboos, 2022-09-13:
I see. Yes, make commands don't run any clippy lints, that's only done via CI at the moment. Sorry to confuse you with that unrelated info. For this you'll just want to pass `--all-features` to the clippy invocation to ensure all crates are included.

tsoutsman, 2022-09-14:
> Thanks @dcampbell24! Does this include all of the crates, even optional ones? (specified using `make full` or by passing `--all-features` to cargo)? I'm traveling at the moment and can't check myself.

The PR CI run succeded, and [it uses `--all-features`](https://github.com/theseus-os/Theseus/blob/4389c3c2518130303058a23ce464258bdd5eb1b4/.github/workflows/check-clippy.yaml#L22).

tsoutsman, 2022-09-16:
Thanks to @dcampbell24, the rest of the passing lints were readded in #638.

amab8901, 2023-01-09:
The following lints are solved now:
* [ ] char_lit_as_u8
* [ ] manual_flatten
* [ ] assign_op_pattern
* [ ] collapsible_else_if
* [ ] collapsible_if
* [ ] collapsible_match
* [ ] comparison_to_empty
* [ ] println_empty_string
* [ ] question_mark
* [ ] toplevel_ref_arg
* [ ] unusual_byte_groupings
* [ ] while_let_on_iterator
* [ ] large_enum_variant

kevinaboos, 2023-01-10:
> The following lints are solved now:

Thanks, I checked 'em off!

kevinaboos, 2023-01-10:
Updated as of #795.

kevinaboos, 2023-01-11:
Updated as of #798.

kevinaboos, 2023-01-13:
Updated as of #806.

kevinaboos, 2023-01-18:
Updated as of #812 and #818

kevinaboos, 2023-01-27:
Updated as of #819

kevinaboos, 2023-01-28:
Thanks to all that helped with this issue!

closed 2023-01-28 by kevinaboos

amab8901, 2023-02-10:
I don't think this issue is finished yet. There are more lints that need to be fixed. I'll be working on it and send more PR:s in the coming future

amab8901, 2023-02-10:
These ones:

<extend>
<summary>cargo clippy</summary>

```
```

<extend>

kevinaboos, 2023-02-10:
ah, interesting. So some of the lints were configured as warnings while others were errors... seemingly the same lint just in different code bases (?). 

I think we can open a new issue to track those since the actual clippy _errors_ were already addressed. If you'd like to continue working on this, I'll let you lead that push (also I don't believe you'd be able to modify this issue anyway, so it's best to open a new one with editing privileges for you).

amab8901, 2023-02-11:
oh, it turns out that I just had to run `git pull` to receive the latest version, and then most of those lints were solved

commit ed747af6 (2023-01-28): Finish addressing all clippy lints (#828)

Fixed the following remaining lints to close #526:
* tabs_in_doc_comments
* unnecessary_lazy_evaluations
* unnecessary_mut_passed
* unused_unit
* upper_case_acronyms
* wrong_self_convention

* Added `make` target to Makefile. Closes #743.

* Removed the list of permitted lints, as all are now addressed.

* Allow both `THESEUS_CONFIG` and `FEATURES` (e.g., `boot_spec`)
  to be used in all make targets, e.g., the new `clippy` target.
commit 012c5517 (2023-01-28): Finish addressing all clippy lints (#828)

Fixed the following remaining lints to close #526:
* tabs_in_doc_comments
* unnecessary_lazy_evaluations
* unnecessary_mut_passed
* unused_unit
* upper_case_acronyms
* wrong_self_convention

* Added `make` target to Makefile. Closes #743.

* Removed the list of permitted lints, as all are now addressed.

* Allow both `THESEUS_CONFIG` and `FEATURES` (e.g., `boot_spec`)
  to be used in all make targets, e.g., the new `clippy` target. ed747af67204752cb52b1b0d369a7ff2a1476e6a