theseus-os/Theseus #531: make: No rule to make target libnano_core.a on WSL2

jkugelman, 2022-06-13:
I'm running through the build instructions on WSL2 for the first time and am getting a make error:

```
❯ make
make: *** No rule to make target '/home/jkugelman/Theseus/target/x86_64-theseus/release/libnano_core.a', needed by '/home/jkugelman/Theseus/build/nano_core/nano_core-x86_64.bin'.  Stop.
make: *** Waiting for unfinished jobs....
info: component 'rust-src' is up to date
```

jkugelman, 2022-06-13:
Here's the output from `make -d`:

[make.txt](https://github.com/theseus-os/Theseus/files/8895167/make.txt)

jkugelman, 2022-06-14:
I have `MAKEFLAGS=-j8` in my environment, which enables parallel make with up to 8 jobs. I turned that off and `make` succeeded.

kevinaboos, 2022-06-14:
Thanks for the report. Build parallelism is handled in large part by `cargo`, but apparently we haven't properly set up the inter-target dependencies in the `Makefile`. Oops!

kevinaboos, 2022-06-14:
A good temporary fix for this is to set the built-in `.NOTPARALLEL` option in the `Makefile`.

kevinaboos, 2022-06-14:
A future PR could experiment with _order-only_ dependencies: https://www.gnu.org/software/make/manual/make.html#Prerequisite-Types

But we certainly don't need to bother with that now.

commit 432334df (2022-06-14): Disable parallel `make`, as `cargo` already handles it (#532)

* If `make`-level parallelism is specified, e.g., `MAKEFLAGS=-j8 make` or `make -j8`, a build error occurs. This explicitly disables parallel building disabled until each `make` target prerequisites can be fixed.

* Closes #531.
closed 2022-06-14 by kevinaboos