theseus-os/Theseus #577: Improve mutex performance

tsoutsman, 2022-07-26:
#576 provides a simple mutex implementation; ideally, we'd use a more performant one. Currently, `raw_mutex` only has a fast path (mutex is unlocked) and a slow path (mutex is locked - block current task). We should also have a middle path, which spins for a bit, hoping the mutex is unlocked soon.

Some useful links:
- How Linux does it: https://0xax.gitbooks.io/linux-insides/content/SyncPrim/linux-sync-4.html
- Another possible implementation: https://arxiv.org/pdf/1906.00490.pdf
- The concept of a backoff could be useful in some capacity: https://docs.rs/crossbeam/latest/crossbeam/utils/struct.Backoff.html

amab8901, 2022-12-27:
#576 isn't merged. How can it the provide the simple mutex implementation? Is this issue still relevant?

kevinaboos, 2023-06-05:
@tsoutsman now that we have your new sync primitives, do you want to attempt adding the middle path? or just close this issue?

PR #968 (2023-06-11): Add middle path for blocking mutex
Starting the task ID counters at 1 rather than 0 also lets us represent task IDs using NonZeroUsize permitting niche optimisations e.g. `Option<TaskId>` taking up as much space as a `usize`. This would however involve some unsafe and even if we do decide to implement it, is better left for another PR.

Closes #577.
closed 2023-06-11 by kevinaboos