theseus-os/Theseus #587: Remove patches from `Cargo.toml`

tsoutsman, 2022-08-05:
Our eventual goal is to have `std` depend on `libtheseus`. This means getting rid of the patch tables in `Cargo.toml`, as they only work within a workspace:
- [ ] `spin`

  ```toml
  ### Patch `spin` to use the `pause` asm instruction in busy-wait loops,
  ### because the `core::hint::spin_loop()` only uses it if "sse2" is enabled.
  spin = { git = "https://github.com/theseus-os/spin-rs" }
  ```
- [ ] `volatile`

  ```toml
  ### use our own version of volatile which supports zerocopy
  volatile = { git = "https://github.com/theseus-os/volatile" }
  ```
- [ ] `getopts`

  ```toml
  ### use our own no_std-compatilbe getopts
  getopts = { git = "https://github.com/theseus-os/getopts" }
  ```
- [ ] `smoltcp` - we can probably just depend on `v0.5.0`
  
  ```toml
  ### use the latest version of smoltcp from github; the one on crates.io is out of date
  smoltcp = { git = "https://github.com/m-labs/smoltcp", rev = "0fedb1db9aa26712830822dd61f065deaa34d611" }
  ```
- [ ] `libc` - either upstream or remove entirely

  ```toml
  ### Patch `libc` so we can use libc-specific types when using `cfg(target_os = "theseus")`.
  libc = { path = "ports/libc" }
  ```
- [ ] `core2`

  ```toml
  ### Patch `core2` with newer functions from `std::io`, e.g., additional `Seek` trait functions
  core2 = { path = "libs/core2" }
  ```
- [ ] `bincode` - This will eventually be fixed upstream

  ```toml
  ### Patch `bincode` because the version on crates.io doesn't handle no_std features correctly.
  bincode = { git = "https://github.com/bincode-org/bincode" }
  ```
### Wasmtime-related crates
- [ ] `wasmparser`
- [ ] `backtrace`
- [ ] `region`
- [ ] `thiserror_core2`

kevinaboos, 2022-08-12:
- `spin`:  we can probably just use the upstream directly; that needs some testing though to ensure system perf isn't affected.

- `volatile`:  our fork of `volatile` has diverged significantly from the original crate; we should simply rename that and publish it as a completely different crate, with `zerocopy` being an optional feature/dependency.
 
- `getopts`:  not upstreamable because this project is a rustc dependency. Similar to volatile we could republish it as a new package.
 
- `smoltcp`:  agreed, update to v0.5.0 and/or the latest version on crates.io. An update is needed anyway to bring more key networking features to Theseus in order to support `std::net` components.
 
- `libc`:  can eventually upstream but it's going to be a long road. Ideally we can just skip all dependencies on libc for every Theseus project.
   -  I actually already changed this patch to depend on our fork via github rather than a local path to a git submodule, so that should make it easier on you in `std`.
 
- `core2`:  i can submit a PR for our changes but it's unlikely to be accepted from what I can tell.
 
- `bincode`:  yes, let's wait til v2.0 is published to crates.ios

kevinaboos, 2022-12-07:
`spin` patch was removed in #719