theseus-os/Theseus #604: Linker script error (nano_core binary linkage)

NathanRoyer, 2022-08-16:
My `GNU ld (GNU Binutils) 2.39` complains about our linker script when building Theseus:

```
ld: warning: build/nano_core/asm_common_x86_64.o: missing .note.GNU-stack section implies executable stack
ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
ld: warning: build/nano_core/nano_core-x86_64.bin has a LOAD segment with RWX permissions
```

It appears some sections are either missing or incorrect.

kevinaboos, 2022-08-26:
Looks like we can just add a `.no.GNU-stack` marker section to indicate to the linker that our stack isn't executable. Does that fix the issue on your end?

Also, did you perhaps change some other part of the build procedure? I only ask because typically there isn't a file called `asm_common_x86_64.o`, it's just called `common.o` on my build, which is built from `kernel/nano_core/src/boot/arch_x86_64/common.asm`.

kevinaboos, 2022-08-26:
Actually the changelog from binutils v2.39 implies that we may need to add a `.note.GNU-stack` section to every input file (so, all of the assembly files).

https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob_plain;f=ld/NEWS;hb=refs/heads/binutils-2_39-branch

If you try that does it resolve the error?

NathanRoyer, 2022-08-29:
I appended an empty section with that name to all asm files except `defines.asm` and the first warning disappeared; the second one remain.

kevinaboos, 2022-08-29:
ok great!

For the other issue, i found [this email thread](https://www.mail-archive.com/bug-binutils@gnu.org/msg42058.html) that seems to indicate it might be an issue with the max page size.

If you look at the output of `readelf -lSW` on the `kernel.bin` file you'll see that almost all of the sections are being combined into one loadable segment:
```
Program Headers:
  Type           Offset   VirtAddr           PhysAddr           FileSiz  MemSiz   Flg Align
  LOAD           0x000120 0x0000000000100000 0x0000000000100000 0x000526 0x000526 R E 0x8
  LOAD           0x001000 0xffffffff80101000 0x0000000000101000 0x191500 0x199000 RWE 0x1000
  LOAD           0x000000 0xffffffff80600000 0x0000000000600000 0x000000 0x012000 RW  0x1000
  GNU_STACK      0x000000 0x0000000000000000 0x0000000000000000 0x000000 0x000000 RWE 0x10

 Section to Segment mapping:
  Segment Sections...
   00     .init 
   01     .text .rodata .eh_frame .gcc_except_table .data .bss .page_table 
   02     .stack 
   03     
```

I think that's the source of the issue. If we set max page size to 4K then it probably won't force the non-4K-aligned sections to be in the same segment.