theseus-os/Theseus #654: `rand_chacha` crate fails to build on debug mode

kevinaboos, 2022-10-07:
The `random` crate depends on `rand_chacha`, which apparently fails to build correctly in debug mode.

## Steps to reproduce
```
make run BUILD_MODE=debug
```
You'll see an LLVM error about split operands. 

Note that this doesn't happen in regular release builds, which is odd.

## Related
I fixed a similar LLVM error in the `crc32fast` crate here: https://github.com/srijs/rust-crc32fast/pull/22, which makes me think that this is related to how `rand_chacha` specific its cfg gates for enabling SSE instructions.

tsoutsman, 2022-10-09:
cc rust-random/rand#1260 cryptocorrosion/cryptocorrosion#69

kevinaboos, 2022-10-10:
Thanks @tsoutsman, the follow-up issues look good. Appreciate you digging into that!

PR #691 (2022-11-11): Update `ppv-lite86`
Closes #654.
closed 2022-11-11 by kevinaboos