theseus-os/Theseus #659: Support std?

dcampbell24, 2022-10-11:
Is there a plan to support the std? You could use a lot more libraries if you did. I am guessing you could use a lot of Redox OS's userland. They are currently at [tier 2 support](https://doc.rust-lang.org/beta/rustc/platform-support.html#tier-2). As far as I could tell the only relevant documentation is [here](https://docs.rust-embedded.org/embedonomicon/custom-target.html).

kevinaboos, 2022-10-12:
Yep, it's underway. See our [project tracker](https://github.com/orgs/theseus-os/projects/2) and [this `std` tracker issue](https://github.com/theseus-os/rust/issues/12) for more

As Theseus is a single address space and single privilege level OS without a distinct userspace, it's significantly more difficult to build `std` for it and requires a fair bit of complex hacks with cargo, build-std, and rustc-dep-of-std. More specifically, Theseus doesn't have a separation boundary based on system calls, which represent a standalone leaf in the dependency tree that makes it easy to build `std` in a relatively standalone manner from the rest of the underlying OS (kernel).

We wouldn't be able to use much Redox code, primarily because its recent (a few years ago by now) target spec identifies it as a Unix family system, but also because most of its code expects to execute in a userspace environment with an underlying Unix-like kernel.

dcampbell24, 2022-10-12:
Thanks for the reply. I guess I will leave this item open for now in case others are wondering the same thing.

kevinaboos, 2022-10-12:
Sure, no problem. I should also mention that we're always looking for help on that large effort, so if anyone is interested in contributing to `std` atop Theseus, feel free to reach out.