theseus-os/Theseus #679: APIC timer calibration is inaccurate on KVM --> timeslices period is too short

kevinaboos, 2022-10-31:
The APIC timer calibration is wrong or somehow inaccurate when running on KVM (`make run host=yes`), which results in the timeslice period being VERY short, and effectively just  context switching on all CPUs incessantly. Thus, it looks as if the system makes no progress or is hung, but in reality it's running correctly.

By choosing a fixed, very large APIC LVT timer calibration value (e.g., `cfg(apic_timer_fixed)` setting a value of 0x10000000), the system works perfectly well... 

_Originally posted by @kevinaboos in https://github.com/theseus-os/Theseus/issues/676#issuecomment-1295695879_

amab8901, 2022-12-21:
how can I get started solving this?

kevinaboos, 2023-01-05:
Sorry for the delay, I missed your question. This is a difficult issue to debug as I myself am not sure of the exact cause.

I would start by using a variety of timer sources (PIT, RTC, HPET, TSC) to calibrate the local apic timer to see if there is a huge difference in the calibration value; perhaps we are just making a minor math mistake or there is some other issue stemming from our reliance on PIT. 

I would also review Intel's Local APIC documentation to make sure we're understanding how things work in x2apic mode, and double check that we're not making any mistakes when initializing x2apic registers on each local apic. Sorry I don't have a clearer path forward on this issue; it requires some pretty in-depth knowledge about how Local APIC works on x2apic systems.

PR #838 (2023-02-14): Fix the APIC timer enable/disable procedure
* The problem: the APIC's timer initial count register must be re-written to when re-enabling that timer after it had been previously disabled.
* We store the current timer count when disabling the APIC timer so that we can write that value back to the initial count register the next time we re-enable the APIC timer.

* Also do a bit of refactoring.

* Makes KVM work again, closes #679
closed 2023-02-14 by kevinaboos