theseus-os/Theseus #686: Use structs for all/most PS/2 functions to prevent arbitrary invocation

hecatia-elegua, 2022-11-06:
is it potentially a problem if this is called randomly by another crate? e.g., before the ps/2 port or mouse has been initialized?
If so, we should make it a member method of some PS/2 struct or something else that prevents it from being arbitrarily invoked.

_Originally posted by @kevinaboos in https://github.com/theseus-os/Theseus/pull/667#discussion_r1012377006_

amab8901, 2022-12-21:
is this issue solved?

hecatia-elegua, 2022-12-21:
Nope, I've started working on this but am still not quite sure about a good structure, so I started working on other drivers.

amab8901, 2022-12-21:
When you say "PS/2 functions", are you referring to the functions in [this](https://github.com/theseus-os/Theseus/blob/e8d9542538a387088ec3b108ed4724e7c5122479/kernel/ps2/src/lib.rs) file?

amab8901, 2022-12-21:
why is the crate called `ps2`?

hecatia-elegua, 2022-12-21:
because someone called it that way

amab8901, 2022-12-21:
oh, I found out that ps\2 stands for `personal system\2` , and that PS\2 controller is the hardware component that controls the communication between the computer and the mouse/keyboard

hecatia-elegua, 2022-12-21:
ah, so that's what your question was about 😄 
Yes, it's an older standard which has been "replaced" by USB, but it's really nice for learning a bit about drivers and getting basic input into an OS (USB is far more unwieldy).

amab8901, 2022-12-21:
> ah, so that's what your question was about smile Yes, it's an older standard which has been "replaced" by USB, but it's really nice for learning a bit about drivers and getting basic input into an OS (USB is far more unwieldy).

oh, so we're doing PS\2 now as a temporary placeholder because USB is much harder to implement due to higher complexity?

kevinaboos, 2022-12-22:
PS/2 isn't just a placeholder, it's still used in modern systems (they often emulate PS/2 hardware for legacy compatibility). USB is more common, but we won't remove PS/2 when USB support is complete.

kevinaboos, 2022-12-22:
> Nope, I've started working on this but am still not quite sure about a good structure, so I started working on other drivers.

@hecatia-elegua can you elaborate the problems you're facing? I'd like to help you solve them. Discord is also fine.

hecatia-elegua, 2022-12-22:
> it's still used in modern systems

I thought it's emulated _because_ of legacy compat, so that old OSes still work. I also read it's more performant for BIOS/early system stuff. But I wouldn't remove PS/2 either.

> can you elaborate the problems you're facing

Let's say I have `PS2Controller::new()`, do I ensure that it's only being called once by kinda building a singleton? So, like in pci: https://github.com/theseus-os/Theseus/blob/981c485b99732783397cae68f02860d1a10a96f7/kernel/pci/src/lib.rs#L78-L81

Other than that, I dislike putting the controller into the mouse and keyboard structs, but it works best. Not really relevant (because over-optimization), but if we had another controller besides the 8042 I would probably add a controller trait and put that into the mouse/keyboard structs.

kevinaboos, 2022-12-22:
> Let's say I have `PS2Controller::new()`, do I ensure that it's only being called once by kinda building a singleton? So, like in pci:

That PCI code is mostly there to cache the result of our brute-force. It also doesn't perform any initialization.

I'd say that you don't have to bother forcing the `PS2Controller` to be a singleton, but you certainly can if you want to and if it makes sense (i.e., use a singleton pattern if there can only ever be one PS2 controller per system).
My primary concern with the existing PS2 code is that most functions are "freestanding" and don't require an initialized PS2/keyboard/mouse object to call; obviously you're solving that right now so that's all good.

> Other than that, I dislike putting the controller into the mouse and keyboard structs, but it works best. Not really relevant (because over-optimization), but if we had another controller besides the 8042 I would probably add a controller trait and put that into the mouse/keyboard structs.

I see. I think that design choice is okay for now because there is no other alternative (right?). I'd recommend to move forward with this approach and submit it as a PR so I can see how it looks; that would make it easier to discuss with concrete examples rather than in the abstract.

A controller trait would be fine, but yes we'd want to hold off on that until we actually have another controller that wants to integrate with the mouse/kbd driver.

hecatia-elegua, 2022-12-23:
Thank you for the help, I really appreciate it.

Now I'm nearly done, the only problem being that I need to access mouse/keyboard in their respective interrupt handlers, which means they need to be 'static, which means the controller needs to be 'static... well, have a look: https://github.com/theseus-os/Theseus/pull/765

hecatia-elegua, 2023-01-04:
Done as of #765

kevinaboos, 2023-01-05:
> Done as of #765

Thanks so much for your contributions here!