theseus-os/Theseus #715: Need to count number of senders and receivers

amab8901, 2022-12-05:
We need to add a counter for the number of senders, and another counter for the number of receivers in `async_channel`. I'm making this into a separate issue because I think the counters can be merged independently, before later merging.

Checklist:

- [ Add sender and receiver counts.] 
- [add `Receiver::obtain_sender()` and `Sender::obtain_receiver()` methods to enable channel to recover from disconnection]
- [add test scenarios. See `applications/test_channel`. ]

This is a prerequisite for https://github.com/theseus-os/Theseus/issues/703

kevinaboos, 2023-01-09:
You're typing in the host terminal, which is what Theseus is using to print its log to (over the virtual serial port). You want to type in the graphical terminal that shows up within a QEMU window.

wlxjhyf, 2024-11-24:
> You're typing in the host terminal, which is what Theseus is using to print its log to (over the virtual serial port). You want to type in the graphical terminal that shows up within a QEMU window.

Hi, Kevin!
So there will be two windows here? One is used to display logs after executing make run, and the other is used in qemu to interact with Theseus? After executing make run, it only displays:
`[W] kernel/console/src/lib.rs:53: Currently ignoring inputs on serial port COM1. 
 --> Note: QEMU is forwarding control sequences (like Ctrl+C) to Theseus. To exit QEMU, press Ctrl+A then X.`
So I really don't know how to interact with Theseus, please answer