theseus-os/Theseus #758: Add a benchmarking mode

tsoutsman, 2022-12-20:
Currently, benchmarking fundamental subsystems (e.g. the task subsystem) is almost impossible. It'd be nice to have a configuration option to run a benchmarking application rather than the shell at startup. The mode could minimise the number of tasks spawned before the benchmark is run.

A dedicated mode would also increase the accuracy of existing benchmarks (e.g. mutex contention).

amab8901, 2022-12-21:
does task subsystem refer to [this](https://github.com/theseus-os/Theseus/blob/theseus_main/kernel/exceptions_full/src/lib.rs) crate?

tsoutsman, 2022-12-21:
> Which of the subsystems are fundamental? Is it all the crates in the kernel directory?

Mostly just the task subsystem. It's hard to test because everything is a task, and so any tasks not relevant to the benchmark impact the benchmark. For example, in #759 the benchmark tasks have to be pinned to core 3 so that they don't yield to the `shell` task.

> I think [flamegraph](https://crates.io/crates/flamegraph) seems like a good choice of benchmarking tool. It can help us see at a glance where performance bottlenecks exist in the OS. Do you agree with my recommendation?

Flamegraph relies on `perf` or `dtrace` depending on the OS, neither of which are available on Theseus. Porting them would be non-trivial.

amab8901, 2022-12-21:
I found another benchmarker: [bencher](https://github.com/bluss/bencher). It seems to be very lightweight with minimal dependencies (if any at all). Would this be a good choice?