theseus-os/Theseus #768: Questions about `Building a C cross compiler` documentation

kuraga, 2022-12-24:
Good day!

I have few questions about [Building GCC and Binutils to target Theseus (x86_64-elf)](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#building-gcc-and-binutils-to-target-theseus-x86_64-elf) section of Theseus Book.

### [1. Build a standalone version of GCC & Binutils](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#1-build-a-standalone-version-of-gcc--binutils).

Is this section really need? Our compile options:
```
../binutils-2.35.1/configure --prefix="$PREFIX" --disable-nls --disable-werror
../gcc-10.2.0/configure --prefix="$PREFIX" --disable-nls --enable-languages=c,c++
```
So, we have have almost "GCC from distribution", don't we?
For which options of above do we compile new?

### [2. Build GCC and Binutils again, to cross-target Theseus (x86_64-elf)](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#2-build-gcc-and-binutils-again-to-cross-target-theseus-x86_64-elf)

It is worth to note: we don't need to compile anything if `gcc -v` has a `Target architecture` equals to Theseus documentation's.
As we don't need, do we? Indeed, our options:
```
../gcc-10.2.0/configure --target=$TARGET --prefix="$PREFIX" --disable-nls --enable-languages=c,c++ --without-headers
```
Or do we need it exactly for `--without-headers` option?

Also, see [this comment](https://stackoverflow.com/questions/74185441/how-do-x86-64-pc-linux-gnu-and-x86-64-elf-toolchains-targets-relate?noredirect=1#comment130979868_74185441).

### [3. Re-building GCC without the default red-zone usage](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#3-re-building-gcc-without-the-default-red-zone-usage)

Can we incorporate this step to the previous one?

Thanks!

kevinaboos, 2022-12-28:
Yep, you don't need to build your own toolchain, instead you can simply download one that disables the redzone (in both gcc and libgcc). Rather than provide a link to a prebuilt that is out of our control, may change, or may not exist in the future, we provide DIY instructions that will always work on its own or as a backup if no other prebuilts are available.

Note that compiling and running any C code directly on Theseus is heavily discouraged and is little more than an experimental proof of concept at this point.

kuraga, 2022-12-29:
@kevinaboos , understand, agree, thanks!

But point-by-point, just for my clarification:

1. [Build a standalone version of GCC & Binutils](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#1-build-a-standalone-version-of-gcc--binutils) - after this step, do we have have almost "GCC from distribution", don't we?
2. [Build GCC and Binutils again, to cross-target Theseus (x86_64-elf)](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#2-build-gcc-and-binutils-again-to-cross-target-theseus-x86_64-elf) - already have answered.
3. [Re-building GCC without the default red-zone usage](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#3-re-building-gcc-without-the-default-red-zone-usage) - can we integrate this step into the previous one? (In sense, don't compile twice but point this step's options at the previous one.)
  **Or do we need both (red-zone and no-red-zone) compilers?**

Thanks!

kevinaboos, 2022-12-29:
> 1. [Build a standalone version of GCC & Binutils](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#1-build-a-standalone-version-of-gcc--binutils) - after this step, do we have have almost "GCC from distribution", don't we?

Hmm, sort of? The key difference is that you have both the source and a GCC binary built from that source, which is not something that you typically get from a distributed GCC binary.

> 3. [Re-building GCC without the default red-zone usage](https://www.theseus-os.com/Theseus/book/c/cross_compiler.html#3-re-building-gcc-without-the-default-red-zone-usage) - can we integrate this step into the previous one?
(In sense, don't compile twice but point this step's options at the previous one.)

I am not sure. The original tutorial we followed required two builds, perhaps it has to do with the bootstrapping procedures required to build a compiler. I am not an expert on the GCC architecture so I'm not sure. If you're able to build it just once and achieve both a gcc and libgcc with no redzone usage, that'd be a good improvement to know about.

> Or do we need both (red-zone and no-red-zone) compilers?

Nope, just the no-red-zone one.

kuraga, 2022-12-29:
Thanks very much!

closed 2022-12-29 by kuraga