theseus-os/Theseus #772: Undefined behaviour in aarch64 `logger`

tsoutsman, 2022-12-27:
## 1
```rust
pub static mut LOGGER: Logger = unsafe { MaybeUninit::uninit().assume_init() };
```
From the [`assume_init` docs](https://docs.rs/maybe-uninit/2.0.0/maybe_uninit/union.MaybeUninit.html#method.assume_init):
> It is up to the caller to guarantee that the MaybeUninit<T> really is in an initialized state. Calling this when the content is not yet fully initialized causes immediate undefined behavior.

## 2
```rust
let mutable = unsafe { (self as *const Self).cast_mut().as_mut().unwrap() };
```
From the [Nomicon](https://doc.rust-lang.org/nomicon/transmutes.html):
> - Transmuting an & to &mut is Undefined Behavior. While certain usages may appear safe, note that the Rust optimizer is free to assume that a shared reference won't change through its lifetime and thus such transmutation will run afoul of those assumptions. So:
>     - Transmuting an & to &mut is always Undefined Behavior.
>     - No you can't do it.
>     - No you're not special.

kevinaboos, 2022-12-28:
@NathanRoyer can chime in here.

I think this will likely be removed once we have UEFI support for aarch64, since we plan to merge the code in `/aarch64` into the main repo.

NathanRoyer, 2023-01-02:
Thank you @tsoutsman for noticing this. Fixed in #784.

PR #784 (2023-01-04): Safer logger on aarch64
Fixes #772; new code makes use of `irq_safety` and its `MutexIrqSafe` struct.
Also includes minimal documentation for this code.

It doesn't use `MaybeUninit` anymore; this has been replaced by wrapping the logger in an option, initially `None`, as suggested by @tsoutsman.
commit 6628d275 (2023-01-04): Fix basic logger on aarch64 (#784)

Closes #772
commit 39fd4596 (2023-01-04): Fix basic logger on aarch64 (#784)

Closes #772 6628d2750d5441281b6f23fb3cf98eedea86507e
closed 2023-01-04 by kevinaboos