theseus-os/Theseus #775: "failed to load source for dependency `backtrace`

amab8901, 2022-12-27:
I deleted my fork of Theseus and also deleted my local copy. I started everything from scratch. I ran `git clone` from upstream repository and I get this issue. I also tried the same from my forked repository, and I still get this issue. Any help would be appreciated, thanks. 

```
$ cargo doc
error: failed to load source for dependency `backtrace`

Caused by:
  Unable to update /home/amab/src/Theseus/ports/backtrace

Caused by:
  failed to read `/home/amab/src/Theseus/ports/backtrace/Cargo.toml`

Caused by:
  No such file or directory (os error 2)

```

tsoutsman, 2022-12-27:
`backtrace` is a submodule. You have to run:
```
git submodule update --init --recursive
```

closed 2022-12-27 by tsoutsman

amab8901, 2022-12-27:
thanks a lot @tsoutsman , it works now