theseus-os/Theseus #778: Support single-core machines

tsoutsman, 2022-12-27:
Running `make run QEMU_CPUS=1` currently fails with:
```
Theseus is shutting down, msg: Couldn't allocate frames for the final framebuffer
[E] kernel/nano_core/src/lib.rs:54: Theseus is shutting down, msg: Couldn't allocate frames for the final framebuffer
[T] kernel/panic_wrapper/src/lib.rs:32: at top of panic_wrapper: PanicInfo { payload: Any { .. }, message: Some(Couldn't allocate frames for the final framebuffer), location: Location { file: "kernel/nano_core/src/lib.rs", line: 57, col: 5 }, can_unwind: true }
```

Sometimes booting APs often complicates things, and so it would be nice to have the option to run Theseus with a single core.

kevinaboos, 2022-12-28:
yep, this is a long-running issue with how obnoxious it is to obtain and modify graphics modes using BIOS interrupts on x86. I think with the new bootloader support this will be fixed on its own, but we can certainly track it in an issue.

kevinaboos, 2022-12-29:
Another thought: this will actually be quite easy since we're implementing support for UEFI GOP. We can reuse the upcoming early framebuffer writer even in BIOS boot mode, as there is a field in the multiboot2 header that can be used to request that the bootloader (grub) puts us directly in graphical mode (rather than in VGA text mode).

In fact I already have the code to request that, I just haven't enabled it because we didn't previously have an early text outputter for graphical framebuffers. 
https://github.com/theseus-os/Theseus/blob/9bd84d88afa4397f4dfc0305bb68b1f6ffe32f03/kernel/nano_core/src/asm/bios/multiboot_header.asm#L17-L32

kevinaboos, 2023-03-18:
This is now addressed thanks to the recent support for early graphical framebuffer support as well as propagating bootloader-provided graphical info throughout initialization. As of PR #891, you can fully boot into a single CPU machine as normal.

closed 2023-03-18 by kevinaboos