theseus-os/Theseus #848: Compilation failed for aarch64

kavanmevada, 2023-02-23:
Compiling tls_initializer v0.1.0 (/home/kavan/Theseus/kernel/tls_initializer)
error[E0425]: cannot find value `TPIDR_EL0` in this scope
   --> kernel/tls_initializer/src/lib.rs:299:9
    |
299 |         TPIDR_EL0.set(self.ptr as u64);
    |         ^^^^^^^^^
    |
   ::: /home/kavan/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-a-7.5.0/src/registers/tpidr_el1.rs:31:1
    |
31  | pub const TPIDR_EL1: Reg = Reg {};
    | ------------------------ similarly named constant `TPIDR_EL1` defined here
    |
help: a constant with a similar name exists
    |
299 |         TPIDR_EL1.set(self.ptr as u64);
    |         ~~~~~~~~~
help: consider importing this constant
    |
14  | use cortex_a::registers::TPIDR_EL0;
    |

warning: unused import: `cortex_a::registers::TPIDR_EL1`
  --> kernel/tls_initializer/src/lib.rs:25:5
   |
25 |     cortex_a::registers::TPIDR_EL1,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: unused import: `tock_registers::interfaces::Writeable`
  --> kernel/tls_initializer/src/lib.rs:26:5
   |
26 |     tock_registers::interfaces::Writeable,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0425`.
warning: `tls_initializer` (lib) generated 2 warnings
error: could not compile `tls_initializer` due to previous error; 2 warnings emitted
warning: build failed, waiting for other jobs to finish...
make: *** [Makefile:317: cargo] Error 101
make ARCH=aarch64 FEATURES= CROSS=aarch64-linux-gnu- run -j1  572.77s user 40.00s system 244% cpu 4:10.82 total

kavanmevada, 2023-02-23:
Can I use `TPIDR_EL1` there as suggested ?

kevinaboos, 2023-02-23:
Thanks for identifying this, we made a mistake yesterday in some TLS-related changes. Oops!

commit 5f4c6a61 (2023-02-23): Fix wrong aarch64 import in `tls_initializer` (#850)

Closes #848
commit 834a9f2f (2023-02-23): Fix wrong aarch64 import in `tls_initializer` (#850)

Closes #848 5f4c6a6191375c61123babd4fa7812ff8f7fccf0
closed 2023-02-23 by kevinaboos