theseus-os/Theseus #849: Missing/invalid symbol expected from assembly code "ap_start_realmode"

kavanmevada, 2023-02-23:
```bash
INFO - nano_core(): initialized memory subsystem.
TRACE - state_store initialized.
INFO - nano_core(): initialized state store.
DEBUG - Created namespace directories: "_kernel"
INFO - nano_core(): initialized crate namespace subsystem.
INFO - nano_core(): parsing nano_core crate, please wait ...
DEBUG - parse_nano_core(): trying to load and parse the nano_core file: Path { path: "/namespaces/_kernel/nano_core.serde" }
DEBUG - Parsing nano_core symbol file: size 0x6a626(435750), mapped_pages: MappedPages { page_table_p4: Frame(p0x40020000), pages: AllocatedPages(Page(v0x100002AA000)..=Page(v0x10000314000)), flags: VALID | PAGE_DESCRIPTOR | READ_ONLY | OUTER_SHAREABLE | ACCESSED | _NOT_GLOBAL | _PRIV_EXEC_NEVER | _USER_EXEC_NEVER | NOT_EXECUTABLE | EXCLUSIVE }, text_pages: Mutex { data: MappedPages { page_table_p4: Frame(p0x40020000), pages: AllocatedPages(Page(v0xFFFF80000000)..=Page(v0xFFFF800C0000)), flags: VALID | PAGE_DESCRIPTOR | READ_ONLY | OUTER_SHAREABLE | ACCESSED | _NOT_GLOBAL | EXCLUSIVE }}, rodata_pages: Mutex { data: MappedPages { page_table_p4: Frame(p0x40020000), pages: AllocatedPages(Page(v0xFFFF800C1000)..=Page(v0xFFFF8011B000)), flags: VALID | PAGE_DESCRIPTOR | READ_ONLY | OUTER_SHAREABLE | ACCESSED | _NOT_GLOBAL | _PRIV_EXEC_NEVER | _USER_EXEC_NEVER | NOT_EXECUTABLE | EXCLUSIVE }}, data_pages: Mutex { data: MappedPages { page_table_p4: Frame(p0x40020000), pages: AllocatedPages(Page(v0xFFFF8011C000)..=Page(v0xFFFF8011E000)), flags: VALID | PAGE_DESCRIPTOR | OUTER_SHAREABLE | ACCESSED | _NOT_GLOBAL | _PRIV_EXEC_NEVER | _USER_EXEC_NEVER | NOT_EXECUTABLE | EXCLUSIVE }}
INFO - Finished parsing nano_core crate, added 2793 new symbols.
INFO - nano_core(): finished parsing the nano_core crate, 2793 new symbols.
INFO - Theseus is shutting down, msg: Missing/invalid symbol expected from assembly code "ap_start_realmode"
ERROR - Theseus is shutting down, msg: Missing/invalid symbol expected from assembly code "ap_start_realmode"
ERROR - Halting due to early panic: panicked at 'Missing/invalid symbol expected from assembly code "ap_start_realmode"', kernel/nano_core/src/lib.rs:70:5
```

kavanmevada, 2023-02-23:
Linux rpi4-20220808 6.1.0-5-arm64 #1 SMP Debian 6.1.12-1 (2023-02-15) aarch64 GNU/Linux

kevinaboos, 2023-02-23:
Hi @kavanmevada, thanks for the report! Can you provide the full build command that you ran when this happened? (e.g., `make run`)

Can you also try a clean build `make clean` and then a basic `make run` after that? This problem indicates something went wrong with the build, or some asm code was modified.

kavanmevada, 2023-02-23:
`make ARCH=aarch64 FEATURES= CROSS=aarch64-linux-gnu- run`

kevinaboos, 2023-02-23:
ah ok, thanks. Our aarch64 port is still a very early work-in-progress, so that error message is currently expected behavior because that's as far as we have gotten in the OS early init on aarch64. Within the next few weeks we should be merging in more subsystems that would allow the OS to proceed further into its bootstrap/init procedure.

cc @NathanRoyer if he would like to provide any more comments/context.

kevinaboos, 2023-02-23:
@NathanRoyer we should use an aarch64-specific statement that prints a log msg about aarch64 boot being complete due to it being partially implemented, e.g., like this for UEFI: https://github.com/theseus-os/Theseus/blob/f1bea2bff7f3bf7b97312d299ce21d37fe293557/kernel/captain/src/lib.rs#L142-L145

closed 2023-03-09 by kevinaboos