theseus-os/Theseus #862: Mitigate risk of getting stuck inside Theseus

amab8901, 2023-03-02:
Getting stuck inside Theseus isn't fun. When I successfully started Theseus for the first time, I felt anxious for a few minutes because I didn't know how to go back to my normal OS. It's like I was stuck in there. Luckily enough, after several minutes of trying different exit methods, I saw on the window name that I could press ctrl + alt + G to release grab. But others won't necessarily find it and may instead decide that "ok I'm stuck here. I need to press the power button to shut down the computer altogether because there's no other way out". This poses a significant risk to user experience and user retention. Some people might even be afraid of using this OS for fear of being unable to get out once they got in.

To mitigate this, I propose the following:
- [ ]  add `poweroff` CLI application (I was looking for a command of this nature when I pressed Tab key in an attempt to exit Theseus)
- [ ] When Theseus terminal is started for the first time, print on the initial message that you can use ctrl+alt+G to release grab (I never expected that this sort of information would be found in the window name, and consequently I didn't look for it in the window name and only looked for the answers within the terminal itself)

kevinaboos, 2023-03-02:
This is a QEMU issue and one that we cannot circumvent. There is a section in the readme about "Using QEMU" that covers this; would you say that is inadequate or needs clarification?

The main problem with printing information like that is that it would be incorrect and irrelevant when running Theseus on anything but QEMU, e.g., real hardware or other emulators.

Support for ACPI-based poweroff/shutdown is on our to-do list, but hasn't been high priority. If you'd like to add support for that, or at least doing it in QEMU (which has its own special set of ports to use for that purpose), I'd be more than happy to welcome a PR.

amab8901, 2023-03-02:
> There is a section in the readme about "Using QEMU" that covers this; would you say that is inadequate or needs clarification?

I think the section in README about "Using QEMU" should mention ctrl + alt + G as a means of escaping QEMU. When I sued ctrl+A and X, it didn't work for exiting QEMU

> The main problem with printing information like that is that it would be incorrect and irrelevant when running Theseus on anything but QEMU, e.g., real hardware or other emulators.

Maybe the message can then print instead something like "try `ctrl + alt + G` or some other keyboard combination if you want to exit Theseus"

kevinaboos, 2023-03-03:
> I think the section in README about "Using QEMU" should mention ctrl + alt + G as a means of escaping QEMU.

Not to be rude, but it quite literally says exactly that. 

> Maybe the message can then print instead something like "try `ctrl + alt + G` or some other keyboard combination if you want to exit Theseus"

Unfortunately, that would be incorrect and misleading because if you press `Ctrl + Alt + G` in the terminal window where the other message about using `Ctrl + A, then X` is printed, it will have no effect.

kevinaboos, 2023-03-03:
AFAICT, this is already explicitly addressed in the documentation, so I will close it. But feel free to file a PR if you believe it can be improved.

closed 2023-03-03 by kevinaboos