theseus-os/Theseus #867: Enable terminal to scroll up

amab8901, 2023-03-03:
I'm currently unable to scroll up the terminal. When I go to `/namespaces/_applications` and insert `ls` command, then I'm literally unable to see the initial part of the output and I'm not able to scroll up and see what it says. It's almost like everything gets cut out except the latter part

[image]

kevinaboos, 2023-03-04:
i'm not sure if our latest GUI + PS/2 mouse implementation still supports mouse scrolling; it used to.

Try scrolling with the keyboard using standard shortcuts, e.g., `Shift + PageUp/PageDown`, `Ctrl + Shift + Up/Down`, `Shift + Home/End`, etc.

amab8901, 2023-03-05:
> 

wow that's awesome, I didn't know you could do that :)

hecatia-elegua, 2023-03-22:
Some pointers to solve this:
- [old impl just used bool flags](https://github.com/theseus-os/Theseus/blob/b60889a7869d16fe01b1d1f72fb001f9edf7df59/libs/mouse_data/src/lib.rs#L130)
- [new impl actually passes the value: scroll_movement](https://github.com/theseus-os/Theseus/blob/5bb39baf3d5a5ebc569b951b99b8050928067453/kernel/ps2/src/lib.rs#L1022)
- [window manager part which needs to be adjusted](https://github.com/theseus-os/Theseus/blob/5bb39baf3d5a5ebc569b951b99b8050928067453/kernel/window_manager/src/lib.rs#L343)

I have not found any code actually handling scrolling, though. The wm is getting reworked, right? Then let's wait with this.