theseus-os/Theseus #960: Potential infinite loop bug in ixgbe driver

Hoblovski, 2023-05-12:
Hi Theseus community, I would like to report a potential bug.

The [following lines of code](https://github.com/theseus-os/Theseus/blob/40805a7934710f16f90000f65497d3a61bb7b681/kernel/ixgbe/src/lib.rs#L852) contains a potential infinite loop bug.
```rust
// kernel/ixgbe/src/lib.rs: IxgbeNic::rx_init
            while rxq.rxdctl.read() & RX_Q_ENABLE == 0 {}
```
The bug is triggered when the device misbehaves, and generates faulty register values that never has the receive enable bit.
The driver will be stuck in this loop, resulting in denial of service.

The linux ixgbe driver avoids the problem via a counter bounding the loop, like
```c
// drivers/net/ethernet/intel/ixgbe/ixgbe_main.c

#define IXGBE_MAX_RX_DESC_POLL 10
static void ixgbe_rx_desc_queue_enable(...) {
	int wait_loop = IXGBE_MAX_RX_DESC_POLL;
	// ...
	do {
		usleep_range(1000, 2000);
		rxdctl = IXGBE_READ_REG(hw, IXGBE_RXDCTL(reg_idx));
	} while (--wait_loop && !(rxdctl & IXGBE_RXDCTL_ENABLE));

	if (!wait_loop) {
		e_err(drv, "RXDCTL.ENABLE on Rx queue %d not set within "
		      "the polling period\n", reg_idx);
	}
}
```

Can we confirm this bug? Also it can be fixed simply via a counter like the linux driver.
I've found a few similar cases but this one is representative.

Thank you very much.

kevinaboos, 2023-05-16:
Thanks for the report! I do agree that we need a timeout there.

I believe that @Ramla-I is working on a series of formally-verified device drivers for the NICs in Theseus, so this is likely already in progress or being addressed with a redesign.

In any case, thanks for bringing this up, and feel free to identify the other places in the code you've found such loops in this issue too.

Ramla-I, 2023-06-12:
Yes, this is a bug and should be addressed. We've just been lucky that we've never been stuck in the infinite loop. I'll make sure to address these changes in a future driver PR, but let's keep this issue open till it's resolved.

Hoblovski, 2023-07-04:
Hi @Ramla-I , sorry for disturbing. I wonder if [the latest bugs](https://github.com/theseus-os/Theseus/issues/960#issuecomment-1600378555) mentioned in the last reply can be confirmed. Thank you very much!

Ramla-I, 2023-07-04:
Right, these are both infinite loop bugs. Thanks for pointing them out!