theseus-os/Theseus #1014: Experiment with using `tracing` instead of `log`

kevinaboos, 2023-07-22:
The `log` crate is decent and easy to support, but it lacks some of the nicer features of more powerful logging systems like `tracing`. For example, it'd be nice to have an easier way to disable all log statements in a certain crate or module, or of a certain level within a given file, etc. The structured logging and span functionality all look attractive too.

The `tracing` crate [does support `no_std`](https://docs.rs/tracing/latest/tracing/#crate-feature-flags), but interestingly enough it requires `alloc` support on `no_std`. So that means we can use it in most crates, but not all crates (i.e., those that must run during early bootstrap before heap allocation is initialized).

cocodery, 2023-11-21:
It seems like a lot of dirty work to do. but maybe not a difficult work?
Almost every crate use `log` crate for tracing, right?
Maybe it's a way to get familiar with `Theseus `source code.

noryve-03, 2024-06-08:
Hello. Been playing around with tracing on my local directory. Is there anyone working on this issue @kevinaboos ?

kevinaboos, 2024-06-08:
Nope, as far as I know it's just an idea. Feel free to go for it!

noryve-03, 2024-06-09:
This issue is intimidating as the scale is huge. Any ideas on incremental improvements? I was thinking of first making a PR for the kernel, then another for applications.

kevinaboos, 2024-06-12:
Definitely! I understand the scale is intimidating. I think a good first step is to look at only the `kernel/logger` crate, which acts as a logger backend and implements the required traits from `log`.

The `tracing` crate can supposedly work as a drop-in replacement for `log`, so you should only need to look at our one `logger` crate to make the relevant changes.

Note that there is no difference in how an application vs a kernel crate does logging in Theseus, they currently all go to the same log stream.

noryve-03, 2024-06-18:
Thank you for the answer! But if I change the backend, won't it also be necessary to replace each call to the `log` crate with a call to the `tracing` crate? Maybe I should not be worried about that? Let me first finish my changes on the `kernel/logger` crate, then I will worry about that.

kevinaboos, 2024-06-21:
oh true, you may need to do that. I'm not sure, part of the fun is finding out! :smiley: 

AFAIK, `tracing` is a drop-in replacement for `log` (based on what i've heard), but you may be correct in that it doesn't directly work with crates that use macros from the actual `log` crate itself.