theseus-os/Theseus #1049: Does not get as far as shell prompt on AArch64

egrimley-arm, 2023-09-26:
I'm running `make ARCH=aarch64 iso` in Docker, then `make ARCH=aarch64 orun host=no` outside Docker. In the terminal it gets as far as `drop(): bootstrap_task_cleanup`, same as on Intel, but in the graphical QEMU window the last thing I see is `INFO : about to jump to kernel: ffff80000000`.

For a fair comparison I also did exactly the same with `ARCH=x86_64` instead of `ARCH=aarch64`: then I get `Theseus Terminal Emulator` and can run `ls` and `ps`.

(Sorry if this is expected behaviour. From #702 I got the impression that AArch64 is mostly working.)

kevinaboos, 2023-09-26:
Thanks for the report! We haven't tried using Docker for aarch64 builds, but I can try it out when I return from a conference. In the meantime @NathanRoyer ought to be able to give a brief update on expected behavior on aarch64; I know we're in the midst of some changes that may have temporarily disabled interactive IO.

NathanRoyer, 2023-09-26:
While most of the core blocks of Theseus have been ported to AArch64, the framebuffer (and the rest of the graphics stack) hasn't fully been; the bootloader is responsible for what you see in the graphical QEMU window.

You can start a console via the serial port (by sending any character), but this console is currently limited to statically linked applications.

egrimley-arm, 2023-09-28:
Thanks! I can confirm that the serial port console does work for me: I get the shell history and `cd`, at least!

NathanRoyer, 2023-09-28:
Great, happy to hear that :ok_hand:

kevinaboos, 2023-10-03:
Thanks @egrimley-arm; i do agree that the current state of the aarch64 build is confusing when you see that partially-complete list of displayed logs on the screen. We'll work on that this week to make regular graphical displays work on aarch64 such that you see something more "normal" on the display.

kevinaboos, 2023-10-03:
I'll update this issue once that's done 👍

kevinaboos, 2023-10-08:
Some to-do items for me to get this working:
* Ensure that all apps are being built for aarch64 so we have a regular interactive console
* Get the framebuffer to work and be initialized on aarch64 as it normally does on x86, or simply disable graphical mode on aarch64 since most boards don't have a default framebuffer device like x86 PCs do.

phansel, 2024-08-31:
Running `make ARCH=aarch64 orun` on Asahi Fedora Remix (as of [these fixes](https://github.com/phansel/Theseus/tree/theseus_main)) results in the same QEMU page stuck on `INFO : about to jump to kernel: ffff80000000`.

fabracht, 2024-10-19:
Running make ARCH=aarch64 run on my Apple M2 Max Sonoma 14.6.1 results in the same page stuck on:
INFO : about to jump to kernel: ffff80000000