theseus-os/Theseus #1053: Theseus fails to boot if built on non-English locale

YtvwlD, 2023-10-11:
I tried building and running Theseus on my Ubuntu 22.04 and it built but it failed to boot:
[image]

(the display is on the left, the serial output is on the right)

Changing `boot_spec` to `uefi` has no effect, but using `loadable` instead of `run` makes it boot.

I tried using a different device (with the same setup) and cloned the repository again into a fresh directory, but it always resulted in a build that doesn't boot.

Then, I went through the issues and found #425 and then I tried setting `LC_ALL="en_US.UTF-8" LANG="en_US.UTF-8"` (from `de_DE.UTF-8`) and it actually worked.

I'm not sure what's going on here or how to triage this, but if I can help somehow, I'll try.

kevinaboos, 2023-10-11:
Thanks for the report! Yeah, looks like it's related to #425 indeed, but I have no idea why we need a particular locale for the build system to work.

> I tried setting LC_ALL="en_US.UTF-8" LANG="en_US.UTF-8" (from de_DE.UTF-8) and it actually worked.

How exactly do you set this, as an environment variable before running `make`? If so, perhaps we can just force it in the Makefile as a temporary workaround.

YtvwlD, 2023-10-11:
yes, I'm running `LC_ALL="en_US.UTF-8" LANG="en_US.UTF-8" make run`

kevinaboos, 2023-10-11:
As someone unfamiliar with other locales, would it be problematic for any reason if we just directly set those env vars in our Makefile?

If that's ok, would you be interested in submitting a PR to do that? If not, no worries. I'm happy to make the change myself.

YtvwlD, 2023-10-11:
It would lead to all output being in English (including numbers), but I guess anyone able to understand the README should be fine with this. Also, qemu and cargo aren't even translated. It's mostly make and git that change.

A technical problem could be that the specific locale might not be installed. I've tried `LC_ALL="C.UTF-8"` with a clean clone and it also works. AFAIK this is available on all modern glibc systems.

I'll try to get a PR ready tomorrow. :)

kevinaboos, 2023-10-11:
Ah, interesting. I have never used another locale so I just wanted to be sure it wouldn't have significant (negative) implications for the user if we forced a locale temporarily.

> I guess anyone able to understand the README should be fine with this

Hope so! 😄  It'd be great to have some internationalization support but that's likely best left for a future date where things are more stable. Maybe one day!

> I've tried `LC_ALL="C.UTF-8"` with a clean clone and it also works. AFAIK this is available on all modern glibc systems.
> 
> I'll try to get a PR ready tomorrow. :)

nice, thanks! no rush.

commit dfa1e367 (2023-10-14): Explicitly set locale in Makefile (#1055)

Set environment variable `LANG="C.UTF-8"`, which ensures that
building and running Theseus on non-English locales will succeed.

Closes #1053.
closed 2023-10-14 by kevinaboos