theseus-os/Theseus #1076: make: *** [cargo] Error 101

ShaunLinTW, 2023-11-21:
"make run" in the docker failed with following error:
...
   Compiling wasmi v0.9.1
   Compiling multiple_heaps v0.1.0 (/Theseus/kernel/multiple_heaps)
   Compiling device_manager v0.1.0 (/Theseus/kernel/device_manager)
   Compiling exceptions_full v0.1.0 (/Theseus/kernel/exceptions_full)
   Compiling first_application v0.1.0 (/Theseus/kernel/first_application)
   Compiling simd_personality v0.1.0 (/Theseus/kernel/simd_personality)
   Compiling task_fs v0.1.0 (/Theseus/kernel/task_fs)
   Compiling utf8parse v0.2.1
   Compiling futures-io v0.3.25
   Compiling arrayvec v0.5.2
   Compiling wasi v0.10.0+wasi-snapshot-preview1 (https://github.com/bytecodealliance/wasi?rev=45536ac956a6211e3cff047f36cf19d6da82fd95#45536ac9)
   Compiling tock-registers v0.7.0
   Compiling vte_generate_state_changes v0.1.1
   Compiling cortex-a v7.5.0
   Compiling vte v0.10.1
   Compiling wasi_interpreter v0.1.0 (/Theseus/kernel/wasi_interpreter)
   Compiling futures v0.3.25
   Compiling captain v0.1.0 (/Theseus/kernel/captain)
   Compiling panic_entry v0.1.0 (/Theseus/kernel/panic_entry)
   Compiling nano_core v0.1.0 (/Theseus/kernel/nano_core)
error: failed to run custom build command for `nano_core v0.1.0 (/Theseus/kernel/nano_core)`

Caused by:
  process didn't exit successfully: `/Theseus/target/release/build/nano_core-76d712d7caedb0c7/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=/Theseus/kernel/nano_core/src/asm
  cargo:rerun-if-changed=/Theseus/build/nano_core/compiled_asm/bios

  --- stderr
  /Theseus/kernel/nano_core/src/asm/ap_boot.asm:1: fatal: unable to open include file `defines.asm'
  thread 'main' panicked at kernel/nano_core/build.rs:167:13:
  assertion failed: Command::new("nasm").args(["-f",
                                                  "elf64"]).arg("-i").arg(&include_path).arg("-o").arg(&output_path).arg(file.path()).args(cflags.split(' ')).status().expect("failed to acquire nasm output status").success()
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
Makefile:320: recipe for target 'cargo' failed
make: *** [cargo] Error 101

kevinaboos, 2023-11-21:
Thanks for the report! We don't actually use the Docker build workflow very often, so it seems like it's been neglected for a while.

The key error is that it cannot find `defines.asm`, which indicates there's a problem with our build script for BIOS builds. My first guess is that something is not getting forwarded properly through the Makefile targets, perhaps an environment variable or something.

I'm about to leave on vacation for a couple of weeks, but in the interim perhaps @tsoutsman can help you get things to build properly. If not, you should be able to build correctly on your native host OS.

kevinaboos, 2023-11-21:
Hmm, it might be that we're missing a trailing slash `/` at the end of the `include_path`, as this SO answer indicates that `nasm` handles include path strings as direct literals.

tsoutsman, 2023-11-23:
@ShaunLinTW could you try the `update-docker` branch. It should have a `nasm` version of `2.15`.

ShaunLinTW, 2023-11-27:
@tsoutsman Hi, I tried to switch to `update-docker` branch, and rmake run again, but this time it shows this error:
shaunlintw@docker-desktop:/Theseus$ make run

=================== BUILDING ALL CRATES ===================
	 TARGET: "x86_64-unknown-theseus"
	 KERNEL_PREFIX: "k#"
	 APP_PREFIX: "a#"
	 CFLAGS: ""
	 THESEUS_CONFIG (before build.rs script): ""
THESEUS_CFLAGS='' THESEUS_NANO_CORE_BUILD_DIR='/Theseus/build/nano_core' RUST_TARGET_PATH='/Theseus/cfg' RUSTFLAGS='--emit=obj -C debuginfo=2 -D unused-must-use -Z share-generics=no ' cargo build --release --workspace --features nano_core/bios -Z unstable-options -Z build-std=core,alloc -Z build-std-features=compiler-builtins-mem --target x86_64-unknown-theseus
info: syncing channel updates for 'nightly-2023-10-27-x86_64-unknown-linux-gnu'
info: latest update on 2023-10-27, rust version 1.75.0-nightly (aa1a71e9e 2023-10-26)
info: downloading component 'cargo'
info: downloading component 'clippy'
info: downloading component 'rust-docs'
info: downloading component 'rust-src'
info: downloading component 'rust-std' for 'aarch64-unknown-uefi'
info: downloading component 'rust-std'
info: downloading component 'rust-std' for 'x86_64-unknown-uefi'
info: downloading component 'rustc'
info: downloading component 'rustfmt'
info: installing component 'cargo'
info: installing component 'clippy'
info: installing component 'rust-docs'
info: installing component 'rust-src'
info: installing component 'rust-std' for 'aarch64-unknown-uefi'
info: installing component 'rust-std'
info: installing component 'rust-std' for 'x86_64-unknown-uefi'
info: installing component 'rustc'
info: installing component 'rustfmt'
error: failed to load source for dependency `backtrace`

Caused by:
  Unable to update /Theseus/ports/backtrace

Caused by:
  failed to read `/Theseus/ports/backtrace/Cargo.toml`

Caused by:
  No such file or directory (os error 2)
make: *** [Makefile:330: cargo] Error 101

ShaunLinTW, 2023-11-27:
...
Compiling elf_cls v0.1.0 (/Theseus/tools/elf_cls)
    Finished release [optimized] target(s) in 2m 20s
     Running `tools/elf_cls/target/release/elf_cls x86_64 --dir /Theseus/build/isofiles/modules`
   Compiling getopts v0.2.17
   Compiling grub_cfg_generation v0.1.0 (/Theseus/tools/grub_cfg_generation)
    Finished release [optimized] target(s) in 10.03s
     Running `tools/grub_cfg_generation/target/release/grub_cfg_generation /Theseus/build/isofiles/modules/ -o /Theseus/build/isofiles/boot/grub/grub.cfg`
qemu-system-x86_64 -cdrom /Theseus/build/theseus-x86_64.iso -boot d -no-reboot -no-shutdown -s -serial mon:stdio -serial mon:pty -m 512M  -smp 4 -net none -cpu Broadwell 
char device redirected to /dev/pts/1 (label serial1-base)
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.fma [bit 12]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.pcid [bit 17]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.tsc-deadline [bit 24]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.avx [bit 28]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.f16c [bit 29]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.avx2 [bit 5]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.fma [bit 12]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.pcid [bit 17]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.tsc-deadline [bit 24]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.avx [bit 28]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.f16c [bit 29]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.avx2 [bit 5]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.fma [bit 12]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.pcid [bit 17]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.tsc-deadline [bit 24]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.avx [bit 28]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.f16c [bit 29]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.avx2 [bit 5]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.fma [bit 12]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.pcid [bit 17]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.tsc-deadline [bit 24]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.avx [bit 28]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.f16c [bit 29]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.avx2 [bit 5]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.invpcid [bit 10]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EBX.rdseed [bit 18]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.80000001H:ECX.3dnowprefetch [bit 8]
gtk initialization failed
make: *** [Makefile:994: orun] Error 1
shaunlintw@docker-desktop:/Theseus$

kevinaboos, 2023-11-29:
> gtk initialization failed

Looks like you are perhaps running on WSL/WSL2? If so, you don't have an X server installed or configured correctly. There's also a section about that on our Readme, in the WSL part. Let me know if you are able to run other graphical X applications but not QEMU -- this problem isn't Theseus-related unfortunately.

ShaunLinTW, 2023-11-30:
Ok! Thanks for the suggestion! It's finally worked!

kevinaboos, 2023-11-30:
Great! Would you mind sharing exactly which step/instruction made things work for you? Perhaps I can clarify or highlight it in the Readme.

kevinaboos, 2024-06-28:
Closing as solved.

closed 2024-06-28 by kevinaboos