theseus-os/Theseus #1097: Network management UI: system default proxy all traffic throughout the system

Pantyhose-X, 2024-01-26:
* System proxy: [wireguard TUN](https://github.com/WireGuard/wireguard-go/tree/master/tun)
[Universal TUN/TAP driver](https://vtun.info/)
[Wintun](https://www.wintun.net/)
https://www.kernel.org/doc/html/v6.1/networking/tuntap.html
https://man.freebsd.org/cgi/man.cgi?query=tun
Network management UI added option TUN 
TUN acts as an agent for existing agreements, HTTP proxy HTTPS proxy SOCKS proxy
Use TUN mode to proxy all traffic throughout the system, including terminals

### Why this request?

I’m in a country where the internet is very blocked, and upgrading OS or downloading software is a hassle, github gitlab Ping over time 300ms.

In thousands of Linux and BSD distributions, there is no single system desktop environment that proxies all traffic,
I use a proxychains and it’s a terrible proxy for the terminal, very slow.

kevinaboos, 2024-01-26:
Hi @Pantyhose-X, thanks for the comment. Unfortunately I'm not quite sure what you're asking for. Is it something along the lines of wanting the source code for Theseus made available via a proxy or a platform that isn't github or US-based? I have heard of some countries blocking github, so I could understand the desire to mirror it elsewhere.

kevinaboos, 2024-01-26:
Or if this is about running Theseus on a machine that uses a proxy, or using a proxy to redirect Theseus's inbound & outbound traffic, we do have instructions for using a tuntap and other forms of virtual networking beneath Theseus in the Readme. Most of it is Linux-focused, but do let us know if you experience difficulty with that. Been awhile since we used some of those more obscure networking features, so the instructions may not be up to date.

Pantyhose-X, 2024-01-26:
> Hi @Pantyhose-X, thanks for the comment. Unfortunately I'm not quite sure what you're asking for. Is it something along the lines of wanting the source code for Theseus made available via a proxy or [...]

Network management UI: Create TUN/TAP, Proxy all traffic

kevinaboos, 2024-01-29:
Right, you said that already, but I was hoping to clarify. You'd like a UI application that runs atop Theseus for the purpose of creating a network proxy?

Pantyhose-X, 2024-01-31:
> You'd like a UI application that runs atop Theseus for the purpose of creating a network proxy?

Yes